symforce-org/symforce #122: CMake issues while building (add_library usage, SYMFORCE_USE_EXTERNAL_LCM option)

gareth-cross, 2022-05-07:
System info:
CMake: 3.18
Python Version: 3.10 w/ virtualenv
Symforce: v0.3.0 

Wasn't sure whether to create separate issues for these:

### add_library + CMake version
The use of `add_library` in `third_party/skymarshal/CMakeLists.txt` specifies source files, which is a CMake 3.19 feature. The min version of 3.17 is too low, and errors out when run with CMake < 3.19 (at least on my Windows machine w/ version 3.18).

What is the preferred fix here? Raise min cmake to 3.19, or remove the source file? I have a vague notion that specifying the include directory with `target_include_directories` is sufficient, but am not 100% sure.

### SYMFORCE_USE_EXTERNAL_LCM default

The current default for `SYMFORCE_USE_EXTERNAL_LCM` is OFF, which causes the build to fail since the target `symforce_lcmtypes_cpp` is not created. I'm guessing the README possibly just needs updating to include `-DSYMFORCE_USE_EXTERNAL_LCM=YES`?

After resolving these two I can get past the cmake step.

aaron-skydio, 2022-05-07:
> add_library + CMake version

I agree with your vague notion that `target_include_directories` is probably sufficient, I'll just do that and keep it at 3.17.

> SYMFORCE_USE_EXTERNAL_LCM default

This should be `NO`, unless you're building in a larger CMake project that already has a target to build the LCM types symforce needs.  I'm guessing it'd fail at build time if you set this to `ON` without providing `SYMFORCE_LCMTYPES_TARGET`?  If not that'd be interesting.  I see at `skymarshal.cmake:57` we're doing the same `add_library` with sources mentioned above, guessing that is failing which is why it's not creating the target?  We can probably just also remove it there?

gareth-cross, 2022-05-07:
Maybe I have this backwards. I'm looking at this code block:

https://github.com/symforce-org/symforce/blob/e2ac9dcdb0a6489af83f75dc66157b387e1c12d7/CMakeLists.txt#L370

`SYMFORCE_USE_EXTERNAL_LCM` is OFF by default, so the else branch is evaluated and the target `symforce_lcmtypes_cpp` does not exist (since it is defined with `add_library` on the non-evaluated branch, like 371).

Something has to define the `symforce_lcmtypes_cpp` target in the case where `SYMFORCE_USE_EXTERNAL_LCM` evaluates to false (the default, I think).

amiller27, 2022-05-07:
Yeah so in the other branch, the call to `add_skymarshal_bindings` on line 376 should call [add_cpp_bindings](https://github.com/symforce-org/symforce/blob/e2ac9dcdb0a6489af83f75dc66157b387e1c12d7/third_party/skymarshal/cmake/skymarshal.cmake#L97), which should create the `symforce_lcmtypes_cpp` target [here](https://github.com/symforce-org/symforce/blob/e2ac9dcdb0a6489af83f75dc66157b387e1c12d7/third_party/skymarshal/cmake/skymarshal.cmake#L57)

aaron-skydio, 2022-05-10:
#127 should fix I think

aaron-skydio, 2022-05-10:
Ok this is actually more annoying than I thought - listing them as sources is necessary so that CMake can track the fact that the library target for these headers depends on the target that actually generates them, so that they're generated before stuff is compiled against them.  It's possible we could rework things to avoid this but it'd be nontrivial - are you locked into cmake 3.18?  I'm going to bump the requirement to 3.19 for now, just curious if there are users that's going to be a problem for

gareth-cross, 2022-05-10:
Yea, that's a fine solution. I only wanted to highlight the inconsistency in the current CMakeLists file. You probably want to update the version in the README instructions as well.

aaron-skydio, 2022-05-10:
Cool, thanks for reporting!

closed 2022-05-16 by aaron-skydio