symforce-org/symforce #161: `pip install -e .` fails on second invocation with "Cython version is too old"

hayk-skydio, 2022-06-05:
**Describe the bug**
On a fresh Python environment I'm able to build SymForce with `pip install -e .`, but calling it again fails.

```
    running build_ext
    SymEngine_DIR : /Users/hayk/projects/symforce/build/temp.macosx-11.1-arm64-cpython-310/symengine_install/lib/cmake/symengine
    SymEngine Version : 0.7.0
    -- Python include path: /Users/hayk/miniconda3/envs/symforceZ/include/python3.10
    -- Python version: 3.10
    -- Python install path: /Users/hayk/projects/symforce/build/temp.macosx-11.1-arm64-cpython-310/symengine_install/lib/python3.10/site-packages
    -- Found CYTHON: /private/var/folders/sd/jqg3nmsn5ns1yzfp9l970xqh0000gn/T/pip-build-env-vmss160g/normal/bin/cython
    CMake Error at cmake/FindCython.cmake:41 (MESSAGE):
      Your Cython version is too old.  Please upgrade Cython.STDOUT: STDERROR:
    Call Stack (most recent call first):
      CMakeLists.txt:17 (find_package)
    
    -- Configuring incomplete, errors occurred!
```

**To Reproduce**
Steps to reproduce the behavior, e.g.:
1. `conda create --name symforce-test "python>=3.8"`
2. `conda activate symforce-test`
3. `pip install -e .`
4. `pip install -e .` again

**Environment (please complete the following information):**
 - OS and version: Mac OS 12.2 M1
 - Python version 3.10.4
 - SymForce Version 3af09b8

gcross-zipline, 2022-06-14:
For those that stumble on this issue and want to work around it: Deleting the `build` directory resolved this issue for me.

aaron-skydio, 2022-06-14:
You should also be able to just delete the directory `build/temp.linux-x86_64-cpython-38/symenginepy-prefix/src/symenginepy-build/build/lib.linux-x86_64-cpython-38` within the build directory (replacing linux, x86_64, and/or 38 as necessary) to do less of a rebuild