symforce-org/symforce #163: Derivative of `sympy.Mod` is 0

aaron-skydio, 2022-06-06:
**Describe the bug**
The derivative of `sympy.Mod` is 0, which breaks the README example on the sympy backend (and anything else that relies on `Mod`)

**To Reproduce**
```python
>>> import symforce
>>> symforce.set_backend("sympy")
>>> from symforce import sympy as sm
>>> x = sm.Symbol(x)
>>> sm.Mod(x, 1).diff(x)
0
```

**Expected behavior**
`sm.Mod(x, 1).diff(x)` should be 1

hayk-skydio, 2022-06-07:
[image]

I think plain sympy doesn't have a derivative for mod? But we add `setattr(sympy.Mod, "_eval_derivative", lambda s, v: sympy.S.Zero)` in `symforce/__init__.py`. This could be a bug as it's next to floor and sign, which actually have zero derivatives.

aaron-skydio, 2022-06-07:
Ah yep, that's it, we should move that into `initialization.py`.  I believe the correct derivative is
`sm.Mod(a, b).diff(a) = 1` and `sm.Mod(a, b).diff(b) = -floor(a/b)`?

Since
```
mod(a, b) = a - b * floor(a/b)
```

(although I'm not sure how sympy mod is defined when a and/or b are negative...)

hayk-skydio, 2022-06-07:
That sounds right to me. Seems worth checking the negative behavior and matching that with the derivative (hopefully it matches the floor expression you wrote.

commit 046ad12f (2022-06-07): [SymForce] Fix derivatives of sm.Mod

Fixes #163

Also added unit tests for all our added derivatives
commit f091044c (2022-06-07): [SymForce] Fix derivatives of sm.Mod

Fixes #163

Also added unit tests for all our added derivatives
closed 2022-06-10 by aaron-skydio