symforce-org/symforce #182: ubuntu 20.04 spdlog via apt get_level build error

jmackay2, 2022-06-24:
**Describe the bug**
Spdlog build error on Ubuntu 20.04 when spdlog is installed with apt, which provides spdlog version 1.5.0. Spdlog is sometimes previously installed via apt for other applications (ROS2). The build succeeds on Ubuntu 22.04 since the apt spdlog version is 1.9.2. If spdlog is not installed previously via apt, there is no issue with the build and installation on Ubuntu 20.04.

**To Reproduce**
Steps to reproduce the behavior, e.g.:
1. docker run -it --rm ubuntu:20.04
2. apt update && DEBIAN_FRONTEND=noninteractive apt install python3-pip libgmp-dev libspdlog-dev build-essential cmake git -y && pip3 install cython
3. git clone https://github.com/symforce-org/symforce.git && cd symforce
4. pip3 install -e .
5. Error: 
```
/symforce/symforce/opt/internal/logging_configure.cc: In function ‘bool sym::internal::SetLogLevel(const string&)’:
/symforce/symforce/opt/internal/logging_configure.cc:38:57: error: ‘get_level’ is not a member of ‘spdlog’; did you mean ‘set_level’?
       38 |                   spdlog::level::to_string_view(spdlog::get_level()));
          |                                                         ^~~~~~~~~
          |                                                         set_level
```

**Expected behavior**
Successful build

**Environment (please complete the following information):**
 - OS and version: Ubuntu Linux 20.04
 - Python version: 3.8.10
 - SymForce Version: Main

aaron-skydio, 2022-06-24:
Thanks for reporting!  `get_level` should be in spdlog 1.8+ ([commit](https://github.com/gabime/spdlog/commit/7009727559901901adef58ca27677d2711b9075a)), we just need to

1) verify that symforce builds probably with 1.8 and 1.10
2) add a min version requirement to our find_package for spdlog
3) verify that we indeed fetch our own copy of spdlog and that this works properly 

This might be annoying with multiple copies of spdlog around though?  Worth also looking into how hard it'd be to get symforce to work with spdlog 1.5 or older

jmackay2, 2022-06-25:
1. Without any code modifications, the minimum spdlog version symforce builds with is 1.9.0. All versions above that also build without issue. 
2. PR has been added to set the min version
3. I have verified that the build is successful with these changes. I haven't done further testing beyond building at the moment.

I don't think having multiple copies of spdlog is too much of an issue.

closed 2022-06-28 by hayk-skydio