symforce-org/symforce #191: Dev command issues on v0.5.0

hayk-skydio, 2022-07-03:
**Describe the bug**
I'm having a series of build issues that are confusing me. Perhaps these should be split issues.

**To Reproduce**
1. Check out v0.5.0
2. Run `rm -rf build` && `pip install -ve .`
3. Verify that importing the library works, and the backend is symengine.

Issue 1: `make lint` and `make check_types` give an error:
```
gen/python/build/lib/sym/pose2.py: error: Duplicate module named "sym.pose2" (also at "gen/python/sym/pose2.py")
gen/python/build/lib/sym/pose2.py: note: Are you missing an __init__.py? Alternatively, consider using --exclude to avoid checking one of them.
Found 1 error in 1 file (errors prevented further checking)
make: *** [check_types] Error 2
```

Issue 2: `make test_update` does nothing:
```
make: Nothing to be done for `test_update'.
```

Issue 3: `make docs` fails:
```
File ~/miniconda3/envs/symforceX/lib/python3.10/site-packages/skymarshal/parser.py:290, in p_array_dim(p)
    286     p[0] = syntax_tree.ArrayDim(tuple())
    287 else:
    288     # Either 1 or 2 words in the brackets, passed through as a tuple.
    289     # May be a literal, a field name, an integer type, or a type and a virtual field name
--> 290     size_declaration = tuple(word for _, word in p[2:-1])
    291     p[0] = syntax_tree.ArrayDim(size_declaration)
    292     p[0].lineno = p[2][0]

File ~/miniconda3/envs/symforceX/lib/python3.10/site-packages/ply/yacc.py:198, in YaccProduction.__getitem__(self, n)
    197 def __getitem__(self,n):
--> 198     if n >= 0: return self.slice[n].value
    199     else: return self.stack[n].value

TypeError: '>=' not supported between instances of 'slice' and 'int'
TypeError: '>=' not supported between instances of 'slice' and 'int'

You can ignore this error by setting the following in conf.py:

    nbsphinx_allow_errors = True
make: *** [docs_html] Error 2
```

**Environment (please complete the following information):**
 - OS and version: `Darwin blitzer.local 21.3.0 Darwin Kernel Version 21.3.0: Wed Jan  5 21:37:58 PST 2022; root:xnu-8019.80.24~20/RELEASE_ARM64_T6000 arm64`
 - Python version `Python 3.10.4`
 - SymForce Version `v0.5.0`

hayk-skydio, 2022-07-03:
Issue 1 looks like it may have been fixed by @bradley-solliday-skydio in #189

aaron-skydio, 2022-07-12:
Agree I think 1 should be fixed by #189 , and the other two should probably be their own issues

2) Hunch is that this is a "macOS `sed` doesn't match linux `sed` thing"?  And that `find test -name "*_codegen_test.py" | sed 's/test\/\(\w\+\).py/update_\1/g'` is empty or something

3) This is seemingly a bug in `ply` that was fixed [in 2013](https://github.com/dabeaz/ply/commit/ee3bb4177786ea343c57208528cc4b5d0cf46918)?  Is your version of `ply` <= 3.8 somehow?

bradley-solliday-skydio, 2022-11-18:
To summarize the state of this issue, issue 1 has been fixed for some time now, and issue 2 was resolved recently (here: https://github.com/symforce-org/symforce/pull/264). Aaron's hypothesis that this had to do with the different behavior of `sed` on macos was correct. Was resolved by updating `sed`'s arguments.

Still need to try reproducing issue 3 with an old version of `ply` to confirm that that is the source of issue. If it is, I'll close the issue on the basis that I think the solution is to simply use a newer version of ply.

aaron-skydio, 2022-11-18:
I don't think the `ply` thing is a symforce issue - if we decide it is because someone hits it again we'll make a separate issue

closed 2022-11-18 by aaron-skydio