symforce-org/symforce #210: cpp version issue

dong-won-shin, 2022-08-09:
Thanks for your great nonlinear optimization library.
I'm testing some features (codegen, sym_opt etc...) and impressed by the improvement in regard to the time complexity.
However, in my application, I need to develop under c++11 so I'm trying to find the way to handle this issue..
I wish to get some advice from your perspective, whether it would be easy or hard.
Again, thanks for your contribution!

aaron-skydio, 2022-08-09:
We definitely aim to be compatible with C++14 and later with no guarantees to be C++11 compatible.  The generated functions themselves probably are C++11 compatible or close to it, but the optimizer would likely require a good amount of modification to compile on C++11 and that's not something we're planning to support.  The generated `sym` types are probably somewhere in between.  Curious if there's a particular platform or compiler you're targeting?

dong-won-shin, 2022-08-09:
I appreciate your valuable comment :-)
In my case, I'm trying to check the feasibility of applying this on the automotive spec.
on this stage, we usually have very limited computational power and memory capacity and low c++ standard.
anyway I found out that the conversion to c++11 would be hard and tedious, that's also a good clue for considering further step.
thank you very much.

closed 2022-08-09 by dong-won-shin