symforce-org/symforce #237: pip install -e . on m1 macbook pro fails to install numpy (but after pip install numpy first, it works)

krosaen, 2022-09-29:
**Describe the bug**

Attempted to install symforce from source following instructions on m1 MacBook Pro, received error.

**To Reproduce**

```
brew install gmp
git clone git@github.com:symforce-org/symforce.git
cd symforce
python3 -m venv env
source env/bin/activate
pip install -e . 
```

and ended up failing:

```
    INFO: CCompilerOpt.cache_flush[857] : write cache to path -> /private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/pip-install-4k39nrum/numpy/build/temp.macosx-10.14-arm64-3.8/ccompiler_opt_cache_clib.py
    error: Command "clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers -arch arm64 -arch x86_64 -Werror=implicit-function-declaration -ftrapping-math -DNPY_INTERNAL_BUILD=1 -DHAVE_NPY_CONFIG_H=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE=1 -D_LARGEFILE64_SOURCE=1 -DNO_ATLAS_INFO=3 -DHAVE_CBLAS -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/src/multiarray -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/src/common -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/src/umath -Inumpy/core/include -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/include/numpy -Ibuild/src.macosx-10.14-arm64-3.8/numpy/distutils/include -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/src/npysort -Inumpy/core/src/common -Inumpy/core/src -Inumpy/core -Inumpy/core/src/npymath -Inumpy/core/src/multiarray -Inumpy/core/src/umath -Inumpy/core/src/npysort -Inumpy/core/src/_simd -I/Users/karlrosaen/local-dev/symforce/env/include -I/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/include/python3.8 -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/src/common -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/src/npymath -c numpy/core/src/multiarray/dragon4.c -o build/temp.macosx-10.14-arm64-3.8/numpy/core/src/multiarray/dragon4.o -MMD -MF build/temp.macosx-10.14-arm64-3.8/numpy/core/src/multiarray/dragon4.o.d -I/System/Library/Frameworks/vecLib.framework/Headers" failed with exit status 1
    ----------------------------------------
    ERROR: Failed building wheel for numpy
  Failed to build cmake numpy
  ERROR: Could not build wheels for cmake, numpy which use PEP 517 and cannot be installed directly
  WARNING: You are using pip version 20.2.3; however, version 22.2.2 is available.
  You should consider upgrading via the '/Users/karlrosaen/local-dev/symforce/env/bin/python3 -m pip install --upgrade pip' command.
  ----------------------------------------
ERROR: Command errored out with exit status 1: /Users/karlrosaen/local-dev/symforce/env/bin/python3 /Users/karlrosaen/local-dev/symforce/env/lib/python3.8/site-packages/pip install --ignore-installed --no-user --prefix /private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/pip-build-env-5eh8k102/normal --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- 'skymarshal @ file://localhost//Users/karlrosaen/local-dev/symforce/third_party/skymarshal' 'cmake>=3.17' pip 'cython>=0.19.1' Check the logs for full command output.
```

**Expected behavior**

Expected it to install properly.

**Environment (please complete the following information):**
 - OS and version: macOS Monterey Version 12.5.1 (21G83) running on Apple M1 Pro
 - Python version 3.8.9
 - SymForce Version `7f9cf250b4647ea73b969a52d70f0de33a9e5e35`

**Additional context**

I was able to complete the install by running `pip install numpy` and then re-running `pip install -e .`

krosaen, 2022-09-29:
I'm also finding that trying to import `symforce.opt.factor` fails with

```
    from symforce.opt.factor import Factor
  File "/Users/karlrosaen/local-dev/symforce/symforce/opt/factor.py", line 23, in <module>
    from symforce.opt.numeric_factor import NumericFactor
  File "/Users/karlrosaen/local-dev/symforce/symforce/opt/numeric_factor.py", line 11, in <module>
    from symforce import cc_sym
  File "/Users/karlrosaen/local-dev/symforce/symforce/cc_sym.py", line 25, in <module>
    from cc_sym import *  # pylint: disable=wildcard-import
ImportError: dlopen(/Users/karlrosaen/local-dev/symforce/cc_sym.cpython-38-darwin.so, 0x0002): Library not loaded: '@rpath/libsymforce_opt.dylib'
  Referenced from: '/Users/karlrosaen/local-dev/symforce/cc_sym.cpython-38-darwin.so'
  Reason: tried: '$ORIGIN/../../libsymforce_opt.dylib' (no such file), '/private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/tmpb61q3xhg.build-temp/libsymforce_opt.dylib' (no such file), '$ORIGIN/../../libsymforce_opt.dylib' (no such file), '/private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/tmpb61q3xhg.build-temp/libsymforce_opt.dylib' (no such file), '/usr/local/lib/libsymforce_opt.dylib' (no such file), '/usr/lib/libsymforce_opt.dylib' (no such file)
```

aaron-skydio, 2022-09-29:
I think this should be fixed by some changes we're publishing later today - in the meantime `pip install .` without `-e` should work correctly

krosaen, 2022-09-29:
Thanks for the update - I ran `pip install .` and it completed, but still get an error importing `Factor`:

```
>>> from symforce.opt.factor import Factor
Traceback (most recent call last):
  File "/Users/karlrosaen/local-dev/symforce/symforce/cc_sym.py", line 18, in <module>
    from cc_sym import *  # pylint: disable=wildcard-import
ImportError: dlopen(/Users/karlrosaen/local-dev/symforce/cc_sym.cpython-38-darwin.so, 0x0002): Library not loaded: '@rpath/libsymforce_opt.dylib'
  Referenced from: '/Users/karlrosaen/local-dev/symforce/cc_sym.cpython-38-darwin.so'
  Reason: tried: '$ORIGIN/../../libsymforce_opt.dylib' (no such file), '/private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/tmpb61q3xhg.build-temp/libsymforce_opt.dylib' (no such file), '$ORIGIN/../../libsymforce_opt.dylib' (no such file), '/private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/tmpb61q3xhg.build-temp/libsymforce_opt.dylib' (no such file), '/usr/local/lib/libsymforce_opt.dylib' (no such file), '/usr/lib/libsymforce_opt.dylib' (no such file)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/karlrosaen/local-dev/symforce/symforce/opt/factor.py", line 23, in <module>
    from symforce.opt.numeric_factor import NumericFactor
  File "/Users/karlrosaen/local-dev/symforce/symforce/opt/numeric_factor.py", line 11, in <module>
    from symforce import cc_sym
  File "/Users/karlrosaen/local-dev/symforce/symforce/cc_sym.py", line 25, in <module>
    from cc_sym import *  # pylint: disable=wildcard-import
ImportError: dlopen(/Users/karlrosaen/local-dev/symforce/cc_sym.cpython-38-darwin.so, 0x0002): Library not loaded: '@rpath/libsymforce_opt.dylib'
  Referenced from: '/Users/karlrosaen/local-dev/symforce/cc_sym.cpython-38-darwin.so'
  Reason: tried: '$ORIGIN/../../libsymforce_opt.dylib' (no such file), '/private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/tmpb61q3xhg.build-temp/libsymforce_opt.dylib' (no such file), '$ORIGIN/../../libsymforce_opt.dylib' (no such file), '/private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/tmpb61q3xhg.build-temp/libsymforce_opt.dylib' (no such file), '/usr/local/lib/libsymforce_opt.dylib' (no such file), '/usr/lib/libsymforce_opt.dylib' (no such file)
```

Next I tried `pip uninstall symforce` and then `pip install .` and ran into

```
 SymEngine Version : 0.7.0
      -- Python include path: /Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers
      -- Python version: 3.8
      -- Python install path: /Users/karlrosaen/local-dev/symforce/build/temp.macosx-10.14-arm64-cpython-38/symengine_install/lib/python3.8/site-packages
      -- Found CYTHON: /private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/pip-build-env-96t_ht_h/normal/bin/cython
      CMake Error at cmake/FindCython.cmake:41 (MESSAGE):
        Your Cython version is too old.  Please upgrade Cython.STDOUT: STDERROR:
      Call Stack (most recent call first):
        CMakeLists.txt:17 (find_package)
```

I couldn't seem to upgrade cython with `pip install --upgrade cython` or by reinstalling it 🤔 

next I tried with a fresh environment and ran into more errors, and can't install `numpy` anymore - so my environment might be hosed in general. Hard to know whether this is symforce specific at this point.

aaron-skydio, 2022-09-29:
The other thing to try along with a fresh environment is making sure the symforce directory is clean, in particular that `build` is deleted/empty - I'll see if I can reproduce some of this on macOS today

krosaen, 2022-09-29:
Ah good point. After removing `build`, starting with a fresh python env, still getting

```
py
    error: Command "clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers -arch arm64 -arch x86_64 -Werror=implicit-function-declaration -ftrapping-math -DNPY_INTERNAL_BUILD=1 -DHAVE_NPY_CONFIG_H=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE=1 -D_LARGEFILE64_SOURCE=1 -DNO_ATLAS_INFO=3 -DHAVE_CBLAS -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/src/multiarray -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/src/common -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/src/umath -Inumpy/core/include -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/include/numpy -Ibuild/src.macosx-10.14-arm64-3.8/numpy/distutils/include -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/src/npysort -Inumpy/core/src/common -Inumpy/core/src -Inumpy/core -Inumpy/core/src/npymath -Inumpy/core/src/multiarray -Inumpy/core/src/umath -Inumpy/core/src/npysort -Inumpy/core/src/_simd -I/Users/karlrosaen/local-dev/symforce/env/include -I/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/include/python3.8 -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/src/common -Ibuild/src.macosx-10.14-arm64-3.8/numpy/core/src/npymath -c numpy/core/src/multiarray/dragon4.c -o build/temp.macosx-10.14-arm64-3.8/numpy/core/src/multiarray/dragon4.o -MMD -MF build/temp.macosx-10.14-arm64-3.8/numpy/core/src/multiarray/dragon4.o.d -I/System/Library/Frameworks/vecLib.framework/Headers" failed with exit status 1
    ----------------------------------------
    ERROR: Failed building wheel for numpy
  Failed to build numpy
  ERROR: Could not build wheels for numpy which use PEP 517 and cannot be installed directly
  WARNING: You are using pip version 20.2.3; however, version 22.2.2 is available.
  You should consider upgrading via the '/Users/karlrosaen/local-dev/symforce/env/bin/python3 -m pip install --upgrade pip' command.
  ----------------------------------------
ERROR: Command errored out with exit status 1: /Users/karlrosaen/local-dev/symforce/env/bin/python3 /Users/karlrosaen/local-dev/symforce/env/lib/python3.8/site-packages/pip install --ignore-installed --no-user --prefix /private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/pip-build-env-cv1y7g6c/normal --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- pip 'cython>=0.19.1' 'skymarshal @ file://localhost//private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/pip-req-build-5fojclqy/third_party/skymarshal' 'cmake>=3.17' Check the logs for full command output.
```

and then I'm still not able to install `numpy` which is weird since it worked earlier.

aaron-skydio, 2022-09-29:
Cool, that's very weird

aaron-skydio, 2022-09-29:
I can't seem to reproduce this, although I'm on a slightly different system:

- macOS 12.5 (21G72), M1 Mac Mini
- Python 3.10 (venv), Python 3.8 (conda)

I'm testing on:
```
commit 63aab44d5c6c2eb674c61447d29df7b063d7720d
Author: Aaron Miller <78561124+aaron-skydio@users.noreply.github.com>
Date:   Thu Sep 29 01:20:08 2022 -0600

    [SymForce-External] Bump version to 0.6
```

which does have some additional install fixes above it that are not yet in `symforce:main`.  I think some of these issues should be fixed by those commits (primarily the `ImportError: dlopen(/Users/karlrosaen/local-dev/symforce/cc_sym.cpython-38-darwin.so, 0x0002): Library not loaded: '@rpath/libsymforce_opt.dylib'`), but I wasn't able to reproduce the numpy thing - my guess is that something else is broken in the environment, and probably the fix isn't symforce-specific but idk if there's something in our install process that can put things into a broken state like that

krosaen, 2022-10-06:
From latest attempt this evening at 693f2702ed5e27a782fae764aea6074c1fae3082 (which is at [release v0.6.0](https://github.com/symforce-org/symforce/releases/tag/v0.6.0) I believe)

```
python3 -m venv env
source env/bin/activate
% python3 -m pip install --upgrade pip
% pip install .                       
Processing /Users/karlrosaen/local-dev/symforce
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Processing //Users/karlrosaen/local-dev/symforce/third_party/skymarshal
  Preparing metadata (setup.py) ... done
Processing //Users/karlrosaen/local-dev/symforce/gen/python
  Preparing metadata (setup.py) ... done
Collecting numpy
  Using cached numpy-1.23.3-cp38-cp38-macosx_11_0_arm64.whl (13.3 MB)
Collecting scipy
  Using cached scipy-1.9.1-cp38-cp38-macosx_12_0_arm64.whl (29.8 MB)
Collecting jinja2
  Using cached Jinja2-3.1.2-py3-none-any.whl (133 kB)
Collecting sympy~=1.11.1
  Downloading sympy-1.11.1-py3-none-any.whl (6.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 6.5/6.5 MB 26.2 MB/s eta 0:00:00
Collecting black
  Using cached black-22.8.0-cp38-cp38-macosx_11_0_arm64.whl (1.2 MB)
Collecting graphviz
  Using cached graphviz-0.20.1-py3-none-any.whl (47 kB)
Collecting clang-format
  Using cached clang_format-14.0.6-py2.py3-none-macosx_10_9_universal2.whl (1.0 MB)
Collecting mpmath>=0.19
  Using cached mpmath-1.2.1-py3-none-any.whl (532 kB)
Collecting pathspec>=0.9.0
  Using cached pathspec-0.10.1-py3-none-any.whl (27 kB)
Collecting mypy-extensions>=0.4.3
  Using cached mypy_extensions-0.4.3-py2.py3-none-any.whl (4.5 kB)
Collecting typing-extensions>=3.10.0.0
  Using cached typing_extensions-4.3.0-py3-none-any.whl (25 kB)
Collecting click>=8.0.0
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Collecting tomli>=1.1.0
  Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
Collecting platformdirs>=2
  Using cached platformdirs-2.5.2-py3-none-any.whl (14 kB)
Collecting MarkupSafe>=2.0
  Using cached MarkupSafe-2.1.1-cp38-cp38-macosx_10_9_universal2.whl (17 kB)
Collecting argh
  Using cached argh-0.26.2-py2.py3-none-any.whl (30 kB)
Collecting ply
  Using cached ply-3.11-py2.py3-none-any.whl (49 kB)
Collecting six
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Using legacy 'setup.py install' for skymarshal, since package 'wheel' is not installed.
Using legacy 'setup.py install' for symforce-sym, since package 'wheel' is not installed.
Building wheels for collected packages: symforce
  Building wheel for symforce (pyproject.toml) ... done
  Created wheel for symforce: filename=symforce-0.6.0-cp38-cp38-macosx_12_0_arm64.whl size=2334975 sha256=6c0ff9cf7c273da46fca37a0c392130697909508b17d9e94c2f897bd87e86069
  Stored in directory: /private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/pip-ephem-wheel-cache-1r9an01b/wheels/9a/9d/a0/a53f589778b118c8c11212333a72a0f63e382943c205d3d929
Successfully built symforce
Installing collected packages: ply, mypy-extensions, mpmath, clang-format, argh, typing-extensions, tomli, sympy, six, platformdirs, pathspec, numpy, MarkupSafe, graphviz, click, symforce-sym, scipy, jinja2, black, skymarshal, symforce
  Running setup.py install for symforce-sym ... done
  Running setup.py install for skymarshal ... done
Successfully installed MarkupSafe-2.1.1 argh-0.26.2 black-22.8.0 clang-format-14.0.6 click-8.1.3 graphviz-0.20.1 jinja2-3.1.2 mpmath-1.2.1 mypy-extensions-0.4.3 numpy-1.23.3 pathspec-0.10.1 platformdirs-2.5.2 ply-3.11 scipy-1.9.1 six-1.16.0 skymarshal-0.6.0 symforce-0.6.0 symforce-sym-0.6.0 sympy-1.11.1 tomli-2.0.1 typing-extensions-4.3.0
```

Looks promising, but then:

```
% python 
[... 23 lines cut ...]
```

as well as

```
>>> from symforce.opt.optimizer import Optimizer
Traceback (most recent call last):
  File "/Users/karlrosaen/local-dev/symforce/symforce/cc_sym.py", line 18, in <module>
    from cc_sym import *  # pylint: disable=wildcard-import
ImportError: dlopen(/Users/karlrosaen/local-dev/symforce/cc_sym.cpython-38-darwin.so, 0x0002): Library not loaded: '@rpath/libsymforce_opt.dylib'
  Referenced from: '/Users/karlrosaen/local-dev/symforce/cc_sym.cpython-38-darwin.so'
  Reason: tried: '$ORIGIN/../../libsymforce_opt.dylib' (no such file), '/private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/tmpb61q3xhg.build-temp/libsymforce_opt.dylib' (no such file), '$ORIGIN/../../libsymforce_opt.dylib' (no such file), '/private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/tmpb61q3xhg.build-temp/libsymforce_opt.dylib' (no such file), '/usr/local/lib/libsymforce_opt.dylib' (no such file), '/usr/lib/libsymforce_opt.dylib' (no such file)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/karlrosaen/local-dev/symforce/symforce/opt/optimizer.py", line 16, in <module>
    from symforce import cc_sym
  File "/Users/karlrosaen/local-dev/symforce/symforce/cc_sym.py", line 25, in <module>
    from cc_sym import *  # pylint: disable=wildcard-import
ImportError: dlopen(/Users/karlrosaen/local-dev/symforce/cc_sym.cpython-38-darwin.so, 0x0002): Library not loaded: '@rpath/libsymforce_opt.dylib'
  Referenced from: '/Users/karlrosaen/local-dev/symforce/cc_sym.cpython-38-darwin.so'
  Reason: tried: '$ORIGIN/../../libsymforce_opt.dylib' (no such file), '/private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/tmpb61q3xhg.build-temp/libsymforce_opt.dylib' (no such file), '$ORIGIN/../../libsymforce_opt.dylib' (no such file), '/private/var/folders/hy/4lcp922d7ml_q264gf67mxzr0000gq/T/tmpb61q3xhg.build-temp/libsymforce_opt.dylib' (no such file), '/usr/local/lib/libsymforce_opt.dylib' (no such file), '/usr/lib/libsymforce_opt.dylib' (no such file)
```

krosaen, 2022-10-06:
I'll try updating to python3.10 to see if that helps.

krosaen, 2022-10-06:
Ok, I do not encounter these issues with python 3.10.7 😌 - updating bug title to reflect 3.8.9 nature.