symforce-org/symforce #239: ModuleNotFoundError: No module named 'lcmtypes.sym' when importing Optimizer

karl-rosaen-skydio, 2022-09-30:
**Describe the bug**
I'm running the code from the main tutorial and get an error during imports.

**To Reproduce**

```
python3.8 -m venv env
source env/bin/activate
pip install -e .
python
>>> import symforce
>>> symforce.get_symbolic_api()
'symengine'
>>> from symforce import cc_sym
```

so far so good, but when typing in the code from the main tutorial and running it, I get an error, which can be reproduced by 

```
>>> from symforce.opt.optimizer import Optimizer
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/karlrosaen/dev/symforce/symforce/opt/optimizer.py", line 11, in <module>
    from lcmtypes.sym._index_entry_t import index_entry_t
ModuleNotFoundError: No module named 'lcmtypes.sym'
```

**Expected behavior**
I'd expect to be able to import the module

**Environment (please complete the following information):**
 - OS and version: Ubuntu 18.04.6
 - Python version 3.8
 - SymForce Version: 7f9cf250b4647ea73b969a52d70f0de33a9e5e35

aaron-skydio, 2022-09-30:
Do you still see this on 48fd72d6244efbf10d06cd65ab24e6ea44c0c7da or later?  I'm not entirely sure if this will be fixed by that or not

aaron-skydio, 2022-10-05:
Confirmed this still exists, and will show up with `pip install -e .` (possibly only on setuptools >=64, not sure).  This is because we install the lcmtypes in a custom addition to the `InstallIWithExtras` command in our setup.py when doing `pip install .`.  For editable installs, this command does not run, but we have duplicated some of its functionality (namely the symengine_wrapper install) into the CMakeBuild command.  We need to also do this for the `lcmtypes.sym` install.

closed 2022-10-19 by aaron-skydio