symforce-org/symforce #246: sym.Pose3.position() read/write data types inconsistency

joansola, 2022-10-11:
**Describe the bug**
Since v0.6.0, the type of `sym.Pose3.t` has changed from a 1d array `ndarray(1,3)` to a vertical array, `ndarray(3,1)` 
Now in v0.6.0, reading Pose3.t produces such a vertical array, but writing into Pose3.t requires a single horizontal array.

**To Reproduce**
```python
pose = sym.Pose3.identity()
position = pose.t
pose2 = sym.Pose3(t=position, R=pose.R)
display(pose2.to_storage())
```
```text
>>>
[0.0, 0.0, 0.0, 1.0, array([0.]), array([0.]), array([0.])]
```

and any operation on `pose2` now fails.

How am I supposed to deal with such positions without going into ugly code such as the one below?
```python
pose = sym.Pose3.identity()
position = pose.t
pose2 = sym.Pose3(t=position.transpose()[0], R=pose.R)
display(pose2.to_storage())
```

**Expected behavior**
```python
pose = sym.Pose3.identity()
position = pose.t
pose2 = sym.Pose3(t=position, R=pose.R)
display(pose2.to_storage())
```
```text
>>>
[0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0]
```

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS and version: Both Ubuntu 20 and MacOS 
 - Python version 3.10
 - SymForce Version 0.6.0 -- as compared to 0.5.0

aaron-skydio, 2022-10-11:
Agree this is quite confusing - we should definitely make these compatible.  Probably what should happen is that the `sym.Pose3` constructor should do the same thing the generated functions do and accept either `T.Union[T.Sequence[float], np.ndarray]` and allow the ndarray to either be flat or a column vector.  No matter what though you should be able to take one `pose.t` and pass it to the constructor of `sym.Pose3`, we should add a unit test that checks this specifically (probably same for the rotation component).

Yes, the workaround for now would be something like that ugly transpose and indexing

joansola, 2022-10-12:
Hi Aaron. I see! Well, what I did, I got rid of all instances of `sym.Pose3` and use instead `sf.Pose3`. Things are a little easier this way and I have less conflicts. The reason why I used `sym` comes back to another old issue you probably remember about copying `Pose3` objects, but this is now solved.

Thanks again! Keep up the good work!

commit 0d8d7ac9 (2022-10-14): sym class constructors accept column vectors

Previously, the class constructors would not accept column vectors.
This is inconvenient because our generated functions accept column
vectors and return column vectors.

So, this PR changes the constructors to accept column vectors as well
(more or less, as row vectors and other funky shapes are also accepted).

Also, replaced assertions on user inputs with exceptions, and asserted
that the user's input vectors, for example, aren't too long, as they
were previously able to silently create wonky Poses (for example.

Notably, this commit does not ch [... cut]
commit fb30d0da (2022-10-15): sym class constructors accept column vectors

Previously, the class constructors would not accept column vectors.
This is inconvenient because our generated functions accept column
vectors and return column vectors.

So, this PR changes the constructors to accept column vectors as well
(more or less, as row vectors and other funky shapes are also accepted).

Also, replaced assertions on user inputs with exceptions, and asserted
that the user's input vectors, for example, aren't too long, as they
were previously able to silently create wonky Poses (for example.

Notably, this commit does not ch [... cut]
commit c281f324 (2022-10-15): sym class constructors accept column vectors

Previously, the class constructors would not accept column vectors.
This is inconvenient because our generated functions accept column
vectors and return column vectors.

So, this PR changes the constructors to accept column vectors as well
(more or less, as row vectors and other funky shapes are also accepted).

Also, replaced assertions on user inputs with exceptions, and asserted
that the user's input vectors, for example, aren't too long, as they
were previously able to silently create wonky Poses (for example.

Notably, this commit does not ch [... cut]
commit 014131af (2022-10-18): sym class constructors accept column vectors

Previously, the class constructors would not accept column vectors.
This is inconvenient because our generated functions accept column
vectors and return column vectors.

So, this PR changes the constructors to accept column vectors as well
(more or less, as row vectors and other funky shapes are also accepted).

Also, replaced assertions on user inputs with exceptions, and asserted
that the user's input vectors, for example, aren't too long, as they
were previously able to silently create wonky Poses (for example.

Notably, this commit does not ch [... cut]
closed 2022-10-21 by aaron-skydio