symforce-org/symforce #256: Add support for Python 3.11

bjacklyn, 2022-10-28:
Python 3.10 works:
```
$ python3.10 -m pip install symforce
Collecting symforce
  Downloading symforce-0.6.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (4.3 MB)
...
```

Python 3.11 doesn't:
```
$ python3.11 -m pip install symforce
ERROR: Could not find a version that satisfies the requirement symforce (from versions: none)
ERROR: No matching distribution found for symforce
```

Looks like latest release [0.6.0](https://pypi.org/project/symforce/0.6.0/) supports python 3.8 to 3.10:
[image]

aaron-skydio, 2022-10-29:
We're working on py3.11 support - in the meantime you're welcome to try building from source with pip as described here: https://github.com/symforce-org/symforce#build-from-source

We haven't tested this on py3.11 either, so no guarantees it'll work (and no guarantees everything in symforce will work even if the install succeeds).  And the "Build with CMake" instructions there won't work out-of-the-box on py3.11 yet since the pinned package versions we use for that aren't yet 3.11 compatible.

aaron-skydio, 2022-11-30:
Fixed by [v0.7](https://github.com/symforce-org/symforce/releases/tag/v0.7.0) and the changes mentioned in the release notes there

closed 2022-11-30 by aaron-skydio