symforce-org/symforce #271: Should add an option to find external metis

chao-qu-skydio, 2022-11-28:
Either this or we need to add its license in the repo for creating conda package.

aaron-skydio, 2022-11-30:
Can we clarify exactly what this means?  Do we just need to add an initial attempt to `find_package(metis <some min version?> QUIET)` [here](https://github.com/symforce-org/symforce/blob/main/symforce/opt/CMakeLists.txt#L88) like we do for other dependencies?  And then have some ability to build wheels (or just a conda package) that dynamically link against `libmetis.so`?  I believe we currently statically link against `libmetis` in wheel builds.

Or do we just need to include a notice in LICENSE that binary distributions of symforce contain METIS with a reference to its license?

I'd be fine with either of these, we should probably have an option to do the first one (at least for cmake builds if not all the piping for wheels) either way?

chao-qu-skydio, 2022-11-30:
Yeah, I meant adding a `find_package(METIS)`. We could use the cmake file from here https://github.com/ceres-solver/ceres-solver/blob/master/cmake/FindMETIS.cmake

aaron-skydio, 2022-11-30:
SGTM, feel free to go for it