symforce-org/symforce #274: Some ideas for improving repository tooling and automation

ameysutavani, 2022-12-01:
Hello all,

I wanted to start a discussion about a few  repository tooling and automation ideas. Primarily helping and improving development experience.

Here are some ideas that I think are worth pursuing:

- Cmake restructuring to break the project down into composable units. Maybe using Conan or other package manager for easier maintenance, easier composability with dependencies.

- Adding a devcontainer to the project for easier developer experience.

- Improving the CI and/or coverage automation.

Eager to hear everyone's thoughts and additions. Especially looking forward to having a pool of these tooling improvement ideas that we can then pursue based on their easy of implementation, value addition and impact.

CC: @hmartiro , @aaron-skydio

aaron-skydio, 2022-12-07:
I think a devcontainer is probably the most self-contained and immediately useful idea - made a issue #276 for that.

There are some additional ideas in existing issues as well, under the [infra tag](https://github.com/symforce-org/symforce/issues?q=is%3Aopen+is%3Aissue+label%3AInfra) (and relatedly the [build tag](https://github.com/symforce-org/symforce/issues?q=is%3Aopen+is%3Aissue+label%3Abuild), although that one has more minor issues and bugs and fewer interesting feature additions).

What are you envisioning in terms of restructuring things?  I'm not really familiar with Conan. I don't think we'd want to add something that would prevent or discourage users from including SymForce in vanilla CMake projects, or projects using some other build system that wraps CMake like colcon, but I'm not sure if any of the ideas you have in mind would do that.

ameysutavani, 2022-12-17:
Hi @aaron-skydio,
Sorry for the late reply. I agree, conan might be a little non popular option. I was thinking of general cmake improvements such as:

1. Splitting the code into more atomic pieces: both in directory structure which aligns with cmake submodules or sub-projects.
2. Adding mechanism where each of these submodules or these smaller building blocks can be compiled, unit-tested atomically.
3. Adding integration tests where multiple of these modules are combined together.

aaron-skydio, 2022-12-19:
Yeah I'm generally on board with those ideas, although would want to talk through the specifics.  I think setting up a devcontainer is a good place to start, and will probably give you some exposure to how the build and dependencies are set up as well, then we can go from there

ameysutavani, 2022-12-20:
Thanks @aaron-skydio . Sounds good.