symforce-org/symforce #283: Marginalization of factors

hanskumarKEF, 2022-12-22:
Would be great to have ability to easily marginalize factors so that we could easily implement fixed lag smoother.

sudo-robot-destroy, 2022-12-23:
I agree that would be great, I've wondered about how to implement a fixed lag smoother but haven't been able to wrap my head around it. Built-in marginalization would be nice.

ZepherusFF, 2023-03-14:
Just stumbled through this thread while reading up about Symforce... I'm confused how this is then used for real-time/online applications (for example, a Skydio drone) if this doesn't exist. I would compare this to having a NonLinearFactorGraph object vs ISAM2 that marginalizes (if we're comparing to GTSAM). 
Surely it's not tractable to have your factor graph grow indefinitely ? Am I missing something ?