symforce-org/symforce #289: could I check the symbolic derivative equation of my residual function?

dong-won-shin, 2023-01-04:
I want to check the result of derivated equation of my residual function to debug it step by step.
Is it possible to see the results from symengine?

dong-won-shin, 2023-01-04:
does jacobian_helpers.py related to the equation?
https://github.com/symforce-org/symforce/blob/main/symforce/jacobian_helpers.py

aaron-skydio, 2023-01-04:
Yeah you can look at the symbolic expressions for the jacobians - assuming you're getting jacobians using `Codegen.with_linearization` or `Codegen.with_jacobians`, yes those call `jacobian_helpers.tangent_jacobians` internally.  You should be able to get the equivalent expression by calling `tangent_jacobians` on your expression yourself.  You could also look at the `outputs` field on the `Codegen` object returned from `with_linearization` or `with_jacobians`, which should include the symbolic expressions for any jacobians added by those methods.

The jacobians are computed with respect to the SO(3) tangent space (technically the coordinate space) at the input element - depending on what exactly you mean by the angle-axis parameterization, what you may be seeing is the difference between what we're computing:
$$\left. \frac{\partial}{\partial v} \left[ r(R.\mathrm{retract}(v)) \right] \right|_{v=0}$$

and this:
$$\left. \frac{\partial}{\partial v} \left[ r(\mathrm{from\\\_tangent}(\mathrm{to\\\_tangent}(R) + v)) \right] \right|_{v=0}$$

dong-won-shin, 2023-01-05:
great! thanks for your answer!
I'll look into the details of ``jacobian_helpers.tangent_jacobians`` function.
thanks!

closed 2023-01-05 by aaron-skydio