symforce-org/symforce #306: Matrix determinant computation

Jaeyoung-Lim, 2023-03-07:
**Is your feature request related to a problem? Please describe.**
It seems like it is not straightforward to compute a determinant of a matrix, or it is not exposed as part of symforce.

It seems like the determinant computation should be available in sympy, which symforce seems to also test against.(e.g. https://github.com/symforce-org/symforce/blob/a1da5b07e287a3130c1fac4e5f1bb14a787ad1b2/third_party/symenginepy/symengine/tests/test_matrices.py#L172)

Would this be just a matter of exposing the sympy API to symforce?

**Describe the solution you'd like**
Be able to compute the determinant of a matrix similar to the `norm()` operation for vectors.

**Additional context**
Computing a determinant can be useful for handling information matrices.
```
import symforce.symbolic as sf

landmark = sf.V3.symbolic("l")
position = sf.V3.symbolic("p")

bearing = (landmark - position) / (landmark - position).norm()

# Jacobian to landmarks
J_theta = bearing.jacobian(landmark)

# Information Matrix
I_theta = J_theta * J_theta.transpose()
D_optimality = I_theta.det()
```

Jaeyoung-Lim, 2023-03-07:
Small update - a workaround is to make the determinant a 1X1 matrix,
```
D_optimality = Matrix([I_theta.mat.det()])
```
But the symforce vectors can no longer be passed to the `subs` but need to provide each key e.g. (`l0`, `l1`, `l2`) to display the value

bresch, 2023-03-07:
Maybe cast it in a `V1` vector?
`H = sf.V1(I_theta.mat.det()).jacobian(position)`

Jaeyoung-Lim, 2023-03-07:
Suggestions from @bresch worked great! Thanks!

closed 2023-03-07 by Jaeyoung-Lim

aaron-skydio, 2023-03-07:
Yeah generally the `.jacobian` method is only present on matrix (and vector) types and not scalars, although we might add it to scalars for consistency so you can do things like this.  Scalars have `.diff` but that will only differentiate with respect to other scalars

> But the symforce vectors can no longer be passed to the subs but need to provide each key e.g. (l0, l1, l2) to display the value
What do you mean here?  Do you have an example?  This maybe sounds like a bug...

Jaeyoung-Lim, 2023-03-07:
@aaron-skydio When I tried to work around the problem by declaring `Matrix` for the determinant, the symforce vectors were no longer being mapped into the values(e.g. vector `l`), and I had to manually map individual components into the `subs()` (e.g. `l0`, l1`, `l2`)

However when I use the workaround by @bresch to add it as a `sf.V1` vector, then everything worked as expected.

aaron-skydio, 2023-03-07:
What was the exact call to `subs`?  E.g. this seems to work as expected for me:
```python
sf.Matrix([D_optimality]).jacobian(position).subs({landmark: sf.V3(1.5, 2.5, 3.5)})
```