symforce-org/symforce #313: robot_3d_localilzation

VineetTambe, 2023-03-17:
**Describe the bug**

running robot_3d_localization.py throws an error:
```
Traceback (most recent call last):
  File "/home/vrex/symForce/symforce/symforce/examples/robot_3d_localization/robot_3d_localization.py", line 425, in <module>
    main()
  File "/home/vrex/symForce/symforce/symforce/examples/robot_3d_localization/robot_3d_localization.py", line 252, in main
    print(f"Num iterations: {len(result.iteration) - 1}")
AttributeError: 'Result' object has no attribute 'iteration'
```
**To Reproduce**
clone the current symforce github repo:

1. Run ```python3 robot_3d_localization.py ```
2. See error

**Expected behavior**
Should running without any errors

**Environment (please complete the following information):**
 - OS and version: Ubuntu Linux 22.04
 - Python version 3.10.6
 - symforce version  0.8.0
 - symforce-sym version 0.8.0
 - sympy  version 1.11.1

**Additional context**
This is a simple problem that can be fixed by using ```result.iteration_stats``` instead of ```result.iterations``` on line 252 of robot_3d_localization.py

Should I raise a pull request for this fix??

aaron-skydio, 2023-03-17:
I'm a little confused by this - SymForce v0.8 has `result.iteration_stats` here, as far as I can tell: https://github.com/symforce-org/symforce/blob/v0.8.0/symforce/examples/robot_3d_localization/robot_3d_localization.py#L252

There is https://github.com/symforce-org/symforce/commit/9266eb54c6bd17408a0c1eb57454a1c3624190e4 which changed the name to `iterations` in main, but I can't find anywhere where it's just `result.iteration`.

I.e. running the v0.8 examples on a v0.8 install should work, and running either the v0.8 examples or the `main` examples should work on an install from `main`

VineetTambe, 2023-03-20:
I did not modify ```robot_3d_localization.py```. 
I installed symforce using pip and cloned the symforce repo from the main branch (this commit [```d1b538c```](https://github.com/symforce-org/symforce/commit/d1b538c8decca807a640c2ef3b6f0caf98efc829)) was the latest commit.

Looks like a

VineetTambe, 2023-03-20:
After checking the git history I found out that:
the ```result.iteration_stats``` was updated to ```result.iterations``` in ```robot_3d_localization.py```  in commit: ```9266eb54``` & revisionId: ```9266eb54c6bd17408a0c1eb57454a1c3624190e4```

aaron-skydio, 2023-03-20:
Ok yeah I was confused by `result.iteration`, as opposed to `result.iterations`, in the original message.  We indeed changed the API in https://github.com/symforce-org/symforce/commit/9266eb54c6bd17408a0c1eb57454a1c3624190e4 - if you've installed symforce v0.8, you should run the examples on the v0.8 tag (i.e. `git checkout v0.8.0`).  If you install SymForce from source you'll be able to run the examples on the latest commit.

closed 2023-04-17 by aaron-skydio