symforce-org/symforce #314: How to intialize sf.Rot3 with a quaternion??

VineetTambe, 2023-03-20:
How do I initialise ```sf.Rot3``` with a quaternion?? 
The tutorial on rotations talks about getting a ```sf.R3``` from all other representation and even specifies that the underlying structure is a quaternion. But how do I initialise a ```sf.Rot3``` structure if I already have quaternion - this is not specified in the tutorials page (Maybe a good idea to add it there?? Or did I just miss it - if that's the case I apologies and can you direct me to that page)

I also dug around a little and found one way to do it 
for a quaternion: ``` q = s + vx i_hat + vy j_hat + vz k_hat```
```sf.Rot3(sf.Quaternion(xyz= np.array([vx,vy,vz]), w = s))```
Is the above correct??

VineetTambe, 2023-03-20:
Thanks for the reply!
The latter part - of having them distinguished is pretty clear and makes sense.

Suggestion:
Can we have Rot3 initialisation from a quaternion visible more readily and on the tutorials page? From the perspective of a first time users of symforce - having that with the tutorials where initialisation from other representations is present would have had been really helpful!

Thanks!

aaron-skydio, 2023-03-20:
Yeah I think having an example there is a good idea

VineetTambe, 2023-03-20:
Should I add one?? Assuming this [comment](https://github.com/symforce-org/symforce/issues/314#issue-1631278827) is correct.

aaron-skydio, 2023-03-20:
Sure!  Really the example should use `sf.V3` instead of `np.array`, but otherwise that's right

VineetTambe, 2023-03-20:
Do I need some access rights to push to a branch so that I can raise a pull request?
If so, then can I be permitted to push a branch to raise a PR?
If not, then what is the procedure that is followed to raise a PR?

aaron-skydio, 2023-03-20:
You should make a fork of the symforce repo and make your changes there, and then make a PR to this repo from your fork: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

VineetTambe, 2023-03-20:
Done!

closed 2023-03-23 by aaron-skydio