symforce-org/symforce #330: error: can't copy 'build/temp.linux-x86_64-cpython-311/symengine_install/lib/python3.11/site-packages/symengine/lib/symengine_wrapper.cpython-311-x86_64-linux-gnu.so': doesn't exist or not a regular file

KOLANICH, 2023-05-22:
**Describe the bug**
The whole `build/temp.linux-x86_64-cpython-311/symengine_install/lib/python3.11` doesn't exist.

`find` shows that the files are in 
`build/temp.linux-x86_64-cpython-311/symenginepy-prefix/src/symenginepy-build/build/lib.linux-x86_64-cpython-311/symengine/lib/symengine_wrapper.cpython-311-x86_64-linux-gnu.so`
and
`build/temp.linux-x86_64-cpython-311/symengine_install/local/lib/python3.11/dist-packages/symengine/lib/symengine_wrapper.cpython-311-x86_64-linux-gnu.so`

**To Reproduce**
Steps to reproduce the behavior, e.g.:
1. `python3 -m build -nwx .`

**Expected behavior**
Should be built.

**Environment (please complete the following information):**
 - OS and version: Debian sid
 - Python version 3.11
 - SymForce Version c3d2f3c97d77e8d9015276cd9ae7b080df646fc3

realSaddy, 2023-06-02:
I am also facing this issue with python310. Did you have any luck getting it built? @KOLANICH

aaron-skydio, 2023-06-02:
Try #342 ?

KOLANICH, 2023-06-03:
>Did you have any luck getting it built?

Yes, I have built it. But I have done it semi-manually, somewhere editing `CMakeCache.txt`, somewhere replacing pieces of files with own code, somewhere hardcoding paths, somewhere using `patchelf` utility. It took me a lot of time to just build this "masterpiece of marvelous engineering" into a wheel that can be installed with `pip` and with which the examples from the `ReadMe` work. And I had to unload X11 and almost all the system services in order just to have enough RAM to build native code part of this package. The mere invocations of the compiler took a lot of time in my system. Since it is such problematic for me to rebuild the package, I'm not going to try #342 unless I really have to do the painful rebuilding process again.

closed 2023-06-07 by aaron-skydio

KOLANICH, 2023-06-08:
Thank you for fixing this. I have **not tested** this yet, but I **hope** that it works.

commit 76c69d3a (2023-06-02): Fix symengine_wrapper search

So that it works on different platforms that e.g. use `local/lib` or
`site-packages`.

We were doing this correctly for editable installs but not non-editable
installs

Longer-term I'm leaning more and more towards requiring building
our copy of symengine separately first, but this is an easy strict
improvement for now

Fixes #330

Topic: sf-wrapper-glob
Reviewers: bradley,nathan,chao,ryan-b
commit 5bc6e9b1 (2023-06-02): Fix symengine_wrapper search

So that it works on different platforms that e.g. use `local/lib` or
`site-packages`.

We were doing this correctly for editable installs but not non-editable
installs

Longer-term I'm leaning more and more towards requiring building
our copy of symengine separately first, but this is an easy strict
improvement for now

Fixes #330

Topic: sf-wrapper-glob
Reviewers: bradley,nathan,chao,ryan-b
commit 71349254 (2023-06-02): Fix symengine_wrapper search

So that it works on different platforms that e.g. use `local/lib` or
`site-packages`.

We were doing this correctly for editable installs but not non-editable
installs

Longer-term I'm leaning more and more towards requiring building
our copy of symengine separately first, but this is an easy strict
improvement for now

Fixes #330

Topic: sf-wrapper-glob
Reviewers: bradley,nathan,chao,ryan-b