symforce-org/symforce #335: Python package metadata contains local paths

KOLANICH, 2023-05-22:
**Describe the bug**
```
Requires-Dist: symforce-sym @ file://localhost/PATH_TO_symforce_CLONED_REPO/gen/python
```

**To Reproduce**
Steps to reproduce the behavior, e.g.:
1. `python3 -m build -nwx`

**Expected behavior**
No such things should be in wheels.

aaron-skydio, 2023-05-22:
We have the `SYMFORCE_REWRITE_LOCAL_DEPENDENCIES` environment variable to control this - when building wheels to distribute, the intention is to set that environment variable, and distribute wheels for `symforce-sym` and `skymarshal` separately.  If that doesn't work for some reason we should fix that.

Agree this isn't ideal, the reason it's set up like this is so that `pip install .` from the SymForce directory "just works".

I'm curious if there's a specific issue this is causing, or if there's a specific proposal to change this in a way that both supports building standalone wheels for these things and one-line from-source installs

aaron-skydio, 2023-05-22:
We aren't supposed to be building tests, examples, or benchmarks when building wheels (benchmarks don't build by default, which is why they aren't explicitly disabled there, although it wouldn't be a bad idea to explicitly disable them also): https://github.com/symforce-org/symforce/blob/c3d2f3c97d77e8d9015276cd9ae7b080df646fc3/setup.py#L114-L119

If we are, that's a bug that we should also fix