symforce-org/symforce #338: New to symforce

BenedictChannn, 2023-05-23:
Hi, I am new to this topic and currently exploring the use of SymForce library and would like some clarification on the use of it.

1) In the example on the robot 3D localization, the generate function generates a C++ code which has faster runtime. How do I then proceed to use this C++ code to solve the localization problem?

2) The generated .h header files all seem to include sym/pose3.h but I can't seem to find the path for pose3.h.

I am still very new to this and would appreciate some help on these. Also, is there anyone that I can contact (email) to gain a deeper understanding on this library and its applications? Thanks and have a great week ahead!

sudo-robot-destroy, 2023-05-23:
This is related to #209

BenedictChannn, 2023-05-24:
Thanks for the prompt response. I managed to solve my second issue.

For the first issue, is anyone from skydio or the community able to roughly explain/ go through the steps of solving the localization problem? 

- What do we do with the generated header files to solve the localization problem in a faster runtime?
- Are we not able to solve the localization problem normally without the code generation portion?

I would appreciate any help on this!

ryan-brott-skydio, 2023-05-25:
> What do we do with the generated header files to solve the localization problem in a faster runtime?

You can use them with the C++ optimization API (e.g., [here](https://github.com/symforce-org/symforce/blob/main/symforce/examples/robot_3d_localization/run_dynamic_size.cc) and [here](https://github.com/symforce-org/symforce/blob/main/symforce/examples/robot_3d_localization/run_fixed_size.cc)).

> Are we not able to solve the localization problem normally without the code generation portion?

Yes, that's what `main()` does in [this file](https://github.com/symforce-org/symforce/blob/c3d2f3c97d77e8d9015276cd9ae7b080df646fc3/symforce/examples/robot_3d_localization/robot_3d_localization.py#L222).

closed 2023-06-01 by aaron-skydio