symforce-org/symforce #340: Issue with compiling run_fixed_size.cc

BenedictChannn, 2023-05-26:
Hi all, I have some issues when trying to run the run_fixed_size.cc

1) In common.h header file there is a line to include the measurements.h header file. However, the compiler keeps telling me that there is no such file or directory. 

#include <symforce/examples/robot_3d_localization/gen/measurements.h>

I then tried to just do #include "./gen/measurements.h" and the above error didn't pop up anymore. However, many more errors popped up as shown below

[image]

2) Has anyone encountered a "TYPE_CHECKING" not defined? I believe it is referenced in the typing.py file and because of the naming convention, the "from typing import *" has some error? I'm not too sure how to resolve the issue.

I would appreciate any help on the above two matters :)

chao-qu-skydio, 2023-05-26:
Can you double-check your fmt and spdlog versions? I see fmt::v6.

BenedictChannn, 2023-06-01:
It was on fmt version 6 yes. I removed and reinstalled using sudo apt-get install libfmt-dev. fmt version is now version 9.1.0 but the issue is still not resolved. Is there an email that I can liaise with on this?

chao-qu-skydio, 2023-06-01:
Could you post the console output?

aaron-skydio, 2023-06-05:
SymForce is not compatible with fmt 6 or fmt 9, we require fmt `8...<9` currently: https://github.com/symforce-org/symforce/blob/main/symforce/opt/CMakeLists.txt#L15

You should build a commit after https://github.com/symforce-org/symforce/commit/f9c21ade57bb920048867d3441efb257e0c7894c if you aren't doing that already (if you're building something after that, I'm confused how the build allowed you to attempt to use fmt 6 at all).  After that commit this should just work, and will use a compatible version of fmt if it's installed, and download it if not

BenedictChannn, 2023-06-07:
I am currently on python 3.8.10.

BenedictChannn, 2023-06-07:
I tried to build from source again after cloning the repo. I attempted cmake .. but the below is what I encounter

cmake ..
-- The C compiler identification is GNU 10.4.0
-- The CXX compiler identification is GNU 10.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Eigen found at /usr/share/eigen3/cmake
-- Catch2 not found, adding with FetchContent
-- Performing Test HAVE_FLAG__ffile_prefix_map__home_ckengjwe_dso_symforce_build__deps_catch2_src__
-- Performing Test HAVE_FLAG__ffile_prefix_map__home_ckengjwe_dso_symforce_build__deps_catch2_src__ - Success
CMake Error at /usr/local/lib/cmake/fmt/fmt-targets.cmake:95 (message):
  The imported target "fmt::fmt" references the file

     "/usr/local/lib/libfmt.a"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/lib/cmake/fmt/fmt-targets.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/local/lib/cmake/fmt/fmt-config.cmake:27 (include)
  symforce/opt/CMakeLists.txt:15 (find_package)

-- Configuring incomplete, errors occurred!

chao-qu-skydio, 2023-06-07:
This seems like you've previously installed fmt and removed it but left the old cmake files? Maybe the find_package is using the old FindFmt.cmake and it's looking for the wrong lib. Just a guess.

BenedictChannn, 2023-06-08:
I managed rebuild somehow and am able to import cc_sym (I was not able to do that before). However, when I tried to compile (I'm not sure if I am doing it correctly) I got the below message:

Starting build...
/usr/bin/cpp -fdiagnostics-color=always -g /home/ckengjwe/dso/symforce/symforce/examples/robot_3d_localization/run_fixed_size.cc -o /home/ckengjwe/dso/symforce/symforce/examples/robot_3d_localization/run_fixed_size
In file included from /usr/local/include/spdlog/common.h:36,
                 from /usr/local/include/spdlog/spdlog.h:12,
                 from /home/ckengjwe/dso/symforce/symforce/examples/robot_3d_localization/run_fixed_size.cc:6:
/usr/local/include/spdlog/fmt/fmt.h:22:14: fatal error: spdlog/fmt/bundled/core.h: No such file or directory
   22 | #    include <spdlog/fmt/bundled/core.h>
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

Build finished with error(s).

 *  The terminal process failed to launch (exit code: -1). 
 *  Terminal will be reused by tasks, press any key to close it. 

It seems to be finding for core.h from /usr/local/include/spdlog but I have checked and core.h is in 
/home/ckengjwe/dso/symforce/build/_deps/spdlog-src/include/spdlog

Somehow the fmt.h file they are using is also from /usr/local/include/spdlog/fmt since it is the default place to search? Is there a way to solve this?

BenedictChannn, 2023-06-08:
Also in run_fixed_size.h header file, I am getting the below error message: 

#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (/home/ckengjwe/dso/symforce/symforce/benchmarks/robot_3d_localization/robot_3d_localization_benchmark.cc).

and I got the below:
[image]

chao-qu-skydio, 2023-06-08:
This seems like you have a slightly messed up environment. If you have fmt or spdlog installed system-wide, then find_package will find them first. It is difficult to help debug your build failure through GitHub issues. May I suggest you start a clean build from a clean environment? For example, you could either remove all system-wide installation of fmt/spdlog or you could do it in a conda environment.

chao-qu-skydio, 2023-06-09:
When you run cmake you need to enable examples (which is ON by default)
cmake .. -DSYMFORCE_BUILD_EXAMPLES=On