symforce-org/symforce #341: No module named 'cc_sym'

BenedictChannn, 2023-06-01:
Built with cmake, following the steps provided in the readme but somehow not able to import cc_sym.

aaron-skydio, 2023-06-01:
You mean `from symforce import cc_sym` fails?  What is the error you get?

BenedictChannn, 2023-06-05:
In Line 18 of cc_sym.py, 

from cc_sym import * # pylint: disable=wildcard-import

ModuleNotFoundError: No module named 'cc_sym'

BenedictChannn, 2023-06-05:
When I built with cmake and executed the make -j $(nproc) command and did run into some errors. There seems to be some issue with my fmt library perhaps.

The error specifically mentions the inability to format an argument of type sym::Key& and std::vector<sym::key_t>&.

Can I ask what is the proper way to install fmt to be compatible with symforce? 

Also, I keep having the 'TYPE_CHECKING' is not defined error. Is anyone also encountering the same issue? And knows how to resolve the issue?

aaron-skydio, 2023-07-12:
This was resolved in discussion at #340

closed 2023-07-12 by aaron-skydio