symforce-org/symforce #353: sf.logical_or

BenedictChannn, 2023-07-12:
```
>>> sf.logical_or(1 > 4, 3 > 2, 8 < 10, unsafe = True)
max(False, True)
```

As long as I have more than 2 arguments, the output is as such. Shouldn't it just output True?

ryan-brott-skydio, 2023-07-12:
`sf.logical_or()` operates on `sf.Scalar`s (`float`s) and not `bool`s. This version of your example works:
```
>>> sf.logical_or(sf.greater(1, 4), sf.greater(3, 2), sf.greater(8, 10), unsafe=True)
1
```

closed 2023-07-12 by ryan-brott-skydio

BenedictChannn, 2023-07-13:
The example I gave above wasn't appropriate... If I am dealing with symbolic variables then even with `sf.greater(x, 5)` for example would result in the same issue. Is there something else I have to do?

ryan-brott-skydio, 2023-07-13:
Can you give some more context on what you're doing with the symbolic expression? In particular, what is the eventual type of `x`? Throwing a symbolic variable in my earlier example seems to give the right result:
```
>>> phi = sf.Symbol('phi')

>>> sf.logical_or(sf.greater(1, 4), sf.greater(3, 2), sf.greater(phi, 10), unsafe=True)
max(1, sign(-10 + phi))
```