symforce-org/symforce #362: problem with code generating

MohammadrezaDindarloo, 2023-08-21:
hello everyone, I faceed with a problem in cpp code generating. 
I have some functions and variables and at the last I calculate my cost function in symbolic. At first I got some incorrect result so I debug all the thing and saw something wired. Assume my cost function in res(x,y,z), that x, y, z are my variables and pass from some functions. when I calculate  res() it gives me a function of x, y and z as expected. now when I give value to my parameters in the begining of code the res() give me a number that replace the value of variables at the end and it is correct (base on some my verifying), the wired thing occures here when I use cpp code and replace the values of my variables. the result is different and incorrect respect to python result. (Also I tried to make my res() base on different variables and getting other the numerical variabled and I fount that the bug occures in defining variabled in a vector (p_platform = sf.Vector3.symbolic("p_init"))). Now I think a part of generating code from this part of variabled have some mistake but I couldn't find the answer. Is it possible? how can I found the problem?

thank you 
m.r.dindarloo

aaron-skydio, 2023-08-22:
Not sure which example you're referring to that you've already found, but here's some of what we have in the docs:

- The homepage has examples of some more symbolic residual functions, if that's the part you're looking for help with: https://symforce.org/index.html#build-an-optimization-problem
- It also shows how to generate a factor into C++: https://symforce.org/index.html#generate-runtime-c-code
- The "custom factor generation" example also shows how to define a symbolic factor and generate it into C++: https://symforce.org/examples/custom_factor_generation/README.html

What part of defining, generating, and using a factor are you looking for help with?

MohammadrezaDindarloo, 2023-08-25:
hello, 
I checked the docs and it was helpful to me. the main part that I was dealing with was in generating a custom factor in order  using in GTSAM factors. At first I write my model with sf. and then tried to create factor that I couldn't find a good resource to do it, finally I ended up with making jacobian with symforce library and add them manually in my factor graph in gtsam and it worked. Now I will try to do this again in symforce by generating my factor from an specific model with define variables and add the output factor to my nonlinearfactorgraph gtsam. 
I should have more study to find the details more because I am new in factor graph and using symforce to generate my factor so if any simple example is developed it will be great to me, but in this level I found this library very powerful in creating the model jacobian (before that I used maple and I coudn't generate the output).

thank you.

aaron-skydio, 2023-08-25:
Cool, glad you were able to figure things out

closed 2023-08-25 by aaron-skydio