symforce-org/symforce #364: Convert gtsam::Rot3 to sym::Rot3

MohammadrezaDindarloo, 2023-09-25:
Hi, 
I want to use some generated cpp code from symforce in my gtsam factors and in my evaluateError function I have a gtsam::Rot3, then in this function I use another function that generated but symforce that have this rotation matrix as input. now when I want to give this rotation matrix to my symforce generated function I should change the type from gtsam::Rot3 to sym::Rot3. I couldn't find any good solution for this. what can I do this for solving my problem? thanks for your reply.

ryan-brott-skydio, 2023-09-25:
I think these functions should just work:
```cpp
sym::Rot3d SymforceFromGtsam(const gtsam::Rot3& gtsam_rot3) {
  return sym::Rot3d(gtsam_rot3.toQuaternion());
}

gtsam::Rot3 GtsamFromSymforce(const sym::Rot3d& geo_rot3) {
  return gtsam::Rot3(geo_rot3.Quaternion());
}
```

MohammadrezaDindarloo, 2023-09-26:
It's true. thank you

closed 2023-09-26 by MohammadrezaDindarloo