symforce-org/symforce #374: Rust codegen support

avsaase, 2023-11-11:
**Is your feature request related to a problem? Please describe.**
Rust is getting more popular in the embedded space and it would be great if symforce could generate efficient Rust code.

**Describe the solution you'd like**
Symforce has a Rust codegen backend. Ideally it would also support adding doc comments to the rust code to make it easier to work with the generated code.

**Describe alternatives you've considered**
Call the generated C++ code from Rust with FFI. I haven't tried this yet but it would certainly be much nicer work with generated Rust code directly.

**Additional context**
I searched for an existing feature request about this but wasn't able to find one.

aaron-skydio, 2023-11-13:
This would be great, and it shouldn't be much work to at least start with a minimal Rust backend.  One decision that's going to come up is what matrix library to use for vector and matrix inputs and outputs in the API - SymForce just needs these as containers, we don't need any operators.  Ideally it would support representing both dense and sparse matrices, just dense might be ok initially but it would be annoying to use a different library for representing sparse matrices.

Also, just noting that the first thing to do here is just codegen support, and a Rust version of the optimizer (either through FFI to the C++ optimizer, or a Rust-native optimizer) would be an additional project.

avsaase, 2023-11-14:
Is the linear algebra really only needed for its types? If so, then I would choose one that adds minimal size to the compiled binary. The most use linear algebra library in rust, `nalgebra` is great but it adds quite a bit of size to the binary which is not great for embedded devices. 

Does the chosen library need to support n-dimensional arrays, or only vector and matrices?

avsaase, 2023-12-10:
I had look through the available linear algebra crates and I think `nalgebra` is the only viable option. All the other libraries I found either are not `no_std` compatible, require an allocator, or don't support arbitrary matrix sizes. `nalgebra` is not the lightest dependency to take on but it's widely used in the Rust ecosystem so it's not unlikely that projects already have a dependency on it, in which case there's not extra code size cost.

aaron-skydio, 2023-12-11:
Nice, thanks for checking!  I think it'd make sense to go ahead with `nalgebra` then

matte1, 2024-01-16:
Has there been any movement here on supporting Rust @avsaase @aaron-skydio? I'd give a +1 to using nalgebra.

aaron-skydio, 2024-01-16:
Nothing currently in progress.  I think the other matrix library we could use is `glam` - it might be nice to make this configurable (this shouldn't be hard).

An example of the codegen part at least is the [CUDA backend](https://github.com/symforce-org/symforce/commit/45b2ce2c4d3b6b3e4dc9b1fb2fdf1255cc509c14), sympy also already has a [Rust printer](https://docs.sympy.org/latest/modules/printing.html#module-sympy.printing.rust) so it might be even less work

matte1, 2024-11-02:
I started working on this [here](https://github.com/symforce-org/symforce/pull/405#pullrequestreview-2411752254). The sympy rust backend doesn't seem all that functional or at least I had to overwrite more than I expected and there are definitely holes still.

This is a very rough pass, but I would love to get something minimal in if possible that gets a foothold for working on a more robust backend.

matte1, 2024-12-25:
Initial support was merged [here](https://github.com/symforce-org/symforce/commit/079830aeb9a755abb687bf7633368d23efe25340). Thanks Aaron!

avsaase, 2025-06-21:
I think I found a bug in the rust code generator. The generated functions return a vector when the python function returns a matrix. I create an issue about it here: https://github.com/symforce-org/symforce/issues/439.