symforce-org/symforce #379: feat: Symbolic integration

ThomasGreenhill, 2024-01-04:
I'd like to use sympy's integrate API, similar to how sympy's diff API is already available through symforce.symbolic.

I'm hoping all that's required is adding a `from sympy import integrate` and a `from symengine import integrate` in `symbolic.py`.

Thanks in advance!

aaron-skydio, 2024-01-04:
`SymEngine` doesn't have `integrate` except for series, as far as I can tell (unless they've added it recently).  You can use `sympy.integrate` while using the SymEngine symbolic API the same way you can for `sympy.solve`, as suggested here: https://github.com/symforce-org/symforce/issues/376#issuecomment-1828514482, or you can use `sf.sympy.integrate` if using the SymPy symbolic API.

This should be easy to add though, like you said it would be as simple as `from sympy import integrate` when on the SymPy symbolic API, and a wrapper that calls the sympy version when on the SymEngine symbolic API, like we do for `limit` in `symbolic.py`.

ThomasGreenhill, 2024-01-04:
Thanks for the quick response! I can definitely just use sympy integrate for the time being :)

closed 2024-01-09 by aaron-skydio

ThomasGreenhill, 2024-01-22:
Thanks a lot for taking care of that, Aaron. Much appreciated.

When will the next symforce release be cut?

aaron-skydio, 2024-01-22:
We're planning to cut a release very soon, probably when numba 0.59 is out (the last of our dependencies to not have py3.12 support)