symforce-org/symforce #381: Returning a `Values` with fixed-size Matrix types results in `VectorXd` in resulting LCM struct

gareth-cross, 2024-01-26:
**Describe the bug**

Not quite sure if this is a bug or expected behavior, but returning a `geo.Matrix3` in a `Values` object results in an LCM type that contains `VectorXd` instead of `Matrix3d`.

**To Reproduce**

Run the following code:
```python
from pathlib import Path

import symforce

symforce.set_epsilon_to_symbol()
symforce.set_symbolic_api("symengine")

from symforce import geo
from symforce import typing as T
from symforce.codegen import Codegen, CppConfig
from symforce.jacobian_helpers import tangent_jacobians_chain_rule
from symforce.values import Values

THIS_DIR = Path(__file__).parent.absolute()

def quat_interpolate(q0_xyzw: geo.V4, q1_xyzw: geo.V4, alpha: T.Scalar) -> Values:
    """A function that would ideally return a struct with two Matrix3d for derivatives."""
    q0 = geo.Rot3.from_storage(q0_xyzw)
    q1 = geo.Rot3.from_storage(q1_xyzw)
    w01 = geo.V3(q0.local_coordinates(q1, epsilon=1.0e-16))
    out = q0.retract(w01 * alpha, epsilon=1.0e-16)
    return Values(
        q_out=geo.V4.from_storage(out.to_storage()),
       # I explicitly convert to M33 here, but that does help.
        D0=geo.M33(tangent_jacobians_chain_rule(expr=out, args=[q0])[0]),
        D1=geo.M33(tangent_jacobians_chain_rule(expr=out, args=[q1])[0])
    )

def main():
    config = CppConfig()
    cg = Codegen.function(func=quat_interpolate, config=config, return_key=None)
    cg.generate_function(output_dir=THIS_DIR / "output" / "quat_interpolation", skip_directory_nesting=True)
```

It produces the following class:
```cpp
class res_t
{
    public:
        ::eigen_lcm::Vector4d q_out;

        ::eigen_lcm::VectorXd D0;

        ::eigen_lcm::VectorXd D1;

...
}
```

**Expected behavior**
Ideally the resulting LCM struct would contain `Vector4d`, and two `Matrix3d` objects for `D0` and `D1`.

**Screenshots**
N/A

**Environment (please complete the following information):**
 - Ubuntu 22.04
 - Python 3.8.16
 - SymForce 0.9.0

**Additional context**
N/a

chao-qu-skydio, 2024-02-02:
Should be fixed upstream.

aaron-skydio, 2024-02-23:
Fixed by https://github.com/symforce-org/symforce/commit/575fd0722b12a7049886d7a848d3ef1b30787536

closed 2024-02-23 by aaron-skydio