symforce-org/symforce #386: Add example for nonlinear MPC from ICRA 2023 workshop

da-phil, 2024-03-04:
Hey guys, thanks for sharing this great tool!
I'm planning to use symforce to solve a nonlinear MPC planning problem and noticed that there has been a [workshop announced for ICRA 2023](https://symforce.org/symforce-workshop) which covers a nonlinear MPC example. I'd be super curious about this and wanted to ask if you could add this example to the repo?

aaron-skydio, 2024-03-04:
The site there was just an initial proposal, not a full workshop, I've archived it for now to avoid confusion.  Unfortunately we don't have an example for nonlinear MPC yet, although we've used symforce internally for nonlinear MPC and other planning and control problems at Skydio.  Hopefully the other examples are enough to get your started.  If you have other questions feel free to ask

closed 2024-03-04 by aaron-skydio

da-phil, 2024-03-04:
Ah, I see, I didn't notice that it was only a proposal. As a lazy human being, I thought a little toy example could help to get to a working implementation faster ;)
Would still appreciate examples for planning problems, such as MPC ;)