symforce-org/symforce #398: Poor accuracy versus Ceres in VSLAM

empty-spacebar, 2024-07-13:
Hi,

I replaced Ceres with Symforce in a graph-based VSLAM backend, result in a bad accuracy. 
Heres are the details:

**Screenshots**
[image]
[image]
[image]

**Environment**
 - Ubuntu Linux 18.04
 - SymForce Version 0.9.0

**Additional context**
1. Mostly Ceres could converge within 10 times itertations, while Symforce can hardly converge within 50 times.
2. Drastic drift happened when use the sym::Optimizer parameters which offical examples use(as the screenshots above shows).
3. I have already tested the validity of the values and factor graph built for the problem.

aaron-skydio, 2024-07-24:
It's hard to say much here without knowing a lot more about what you're doing (what do you mean when you say graph-based VSLAM, what factors are included, what initialization, etc).

It would also be useful to know what optimization parameters are being used with Ceres, or if Ceres is selecting any of those automatically, which it's choosing (which optimizer, which linear solver, inner iterations, etc).

It looks like you've tried the DYNAMIC lambda parameter, which is one thing that can improve convergence in more nonlinear problems like bundle adjustment.

One other thing that would be useful to know is whether the SymForce optimizer converges eventually, and how many iterations it takes, if you increase the iteration limit.

empty-spacebar, 2024-07-24:
> It's hard to say much here without knowing a lot more about what you're doing (what do you mean when you say graph-based VSLAM, what factors are included, what initialization, etc).

THe graph-based VSLAM framwork is VINS-MONO, feature-based frontend, factor-graph backend optimized with Gauss–Newton originally, and its factor include poses in quaternion, inverse-depth of landmark points.

> It would also be useful to know what optimization parameters are being used with Ceres, or if Ceres is selecting any of those automatically, which it's choosing (which optimizer, which linear solver, inner iterations, etc).

Here is the CERES parameter that works:
options.max_num_iterations = 8
options.linear_solver_type = ceres::DENSE_SCHUR;
options.trust_region_strategy_type = ceres::DOGLEG

Symforce doesn't have DOGLEG, so I replace ceres::DOGLEG with:
```
using DenseOptimizer =
    sym::Optimizer<Scalar, sym::LevenbergMarquardtSolver<Scalar, sym::DenseCholeskySolver<Scalar>>>; 
```
and the default sparse decomposition:
`sym::Optimizer<double> optimizer`
Here comes the question----Will Symforce add support of more decomposition method like dogleg?

> It looks like you've tried the DYNAMIC lambda parameter, which is one thing that can improve convergence in more nonlinear problems like bundle adjustment.

DYNAMIC lambda parameter doesn't work----with it VINS runs even worse. Thus static lambda is used as the 2nd image above.

> One other thing that would be useful to know is whether the SymForce optimizer converges eventually, and how many iterations it takes, if you increase the iteration limit.

As for  whether the SymForce optimizer converges eventually, it's showed in the fourth image above----seldom converges, usually disconverges and ends up with hitting iteration limit(50 times iterations).

aaron-skydio, 2024-07-24:
Again I'm not sure how much we're going to be able to debug this over GitHub issues, so this may be a dead end, but a couple more thoughts:

> Here is the Ceres parameter that works well in VINS

Is Ceres using inner iterations?  It might help to have the Ceres solver output, e.g. `ceres::Solver::Summary::FullReport()`.  It might be useful to run Ceres with LM and no inner iterations, which would be closer to the SymForce optimizer.

> DYNAMIC lambda parameter doesn't work----with it VINS runs even worse. Thus static lambda is used as the 2nd image above.

I'm not too shocked by this, but I'll note that DYNAMIC is basically what Ceres does by default in their LM implementation, while STATIC is not

> As for whether the SymForce optimizer converges eventually, it's showed in the fourth image above----seldom converges, usually disconverges and ends up with hitting iteration limit(50 times iterations).

What I was asking is, if you set the SymForce iteration limit much higher, what happens?  It's much more useful to know what SymForce would do with an unlimited number of iterations (does it converge in 60?  100?  1000?  or get stuck?) (does optimizing to convergence give the same solution as Ceres?) than knowing that it often takes more than 50.

It is often also useful to see the full verbose optimization trace (by setting the `verbose` parameter to `true`), to know whether the optimizer is making good progress, and if not why not.  Separately, it can be useful to visualize the values at intermediate steps of the optimization (you can save the Values for each iteration in the OptimizationStats by turning on `debug_stats`, and you seem to have some visualization tools set up already that you could probably use for this)

empty-spacebar, 2024-07-26:
> Is Ceres using inner iterations? It might help to have the Ceres solver output, e.g. `ceres::Solver::Summary::FullReport()

I didn't set the option 'use_inner_iterations', and since its default value is false, I guess inner iterations is OFF in my cases.
I pack all my test log up, I hope the file names are clear enough to know where they came from.  Also I took the screenshots of the trajectories of every test.
[symforce_test_20240726.zip](https://github.com/user-attachments/files/16391065/symforce_test_20240726.zip)

> What I was asking is, if you set the SymForce iteration limit much higher, what happens? It's much more useful to know what SymForce would do with an unlimited number of iterations (does it converge [...]

Log of this test is also being record into the file above. It's interesting that when Ceres runs well, it doesn't converge either(mostly  in 4 times iterations, end because iteration time out).
The reason why symforce runs worse is because its residual decreases VERY SLOWLY, sometimes takes 50 times iteration to get a good result(see sym_LM_dense_cholesky.png), sometimes just crush.

And more interesting is, Ceres doesn't run well everytimes(see Ceres_LM_dense_cholesky1~3, which means exactly the same parameter but different results)

Here are some other discoveries:
1, Ceres runs awfully with LM, no matter dense schur or dense cholesky is used;
2, Ceres with LM and sparse cholesky, reports error like 'dense matrix decomposition fail' but still get right trajectory;
3, symforce with LM and sparse cholesky runs bad, with dense cholesky and 50 times iterations runs good.
4, When set lambda zero, the program report:'Internal Error: Damped hessian factorization failed.'

aaron-skydio, 2024-07-26:
That's actually pretty helpful, thanks

Optimization traces like this are exactly those that should be improved by dynamic lambda:
```
[2024-07-26 14:11:26.634] [info] LM<sym::Optimize> [iter    0] lambda: 1.000e+00, error prev/linear/new: 2.010e+02/2.009e+02/2.010e+02, rel reduction: 2.45473e-04, gain ratio: 5.14165e-01
[2024-07-26 14:11:26.636] [info] LM<sym::Optimize> [iter    1] lambda: 1.000e-01, error prev/linear/new: 2.010e+02/2.008e+02/2.011e+02, rel reduction: -4.78170e-04, gain ratio: -6.30568e-01
[2024-07-26 14:11:26.639] [info] LM<sym::Optimize> [iter    2] lambda: 1.000e+00, error prev/linear/new: 2.010e+02/2.009e+02/2.009e+02, rel reduction: 1.02077e-04, gain ratio: 3.61970e-01
[2024-07-26 14:11:26.641] [info] LM<sym::Optimize> [iter    3] lambda: 1.000e-01, error prev/linear/new: 2.009e+02/2.008e+02/2.011e+02, rel reduction: -6.08975e-04, gain ratio: -1.06610e+00
[2024-07-26 14:11:26.643] [info] LM<sym::Optimize> [iter    4] lambda: 1.000e+00, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: 4.45069e-05, gain ratio: 2.27182e-01
[2024-07-26 14:11:26.645] [info] LM<sym::Optimize> [iter    5] lambda: 1.000e-01, error prev/linear/new: 2.009e+02/2.008e+02/2.011e+02, rel reduction: -6.62745e-04, gain ratio: -1.44154e+00
[2024-07-26 14:11:26.647] [info] LM<sym::Optimize> [iter    6] lambda: 1.000e+00, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: 1.60077e-05, gain ratio: 1.08418e-01
[2024-07-26 14:11:26.650] [info] LM<sym::Optimize> [iter    7] lambda: 1.000e-01, error prev/linear/new: 2.009e+02/2.008e+02/2.011e+02, rel reduction: -6.79496e-04, gain ratio: -1.75608e+00
[2024-07-26 14:11:26.652] [info] LM<sym::Optimize> [iter    8] lambda: 1.000e+00, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: 1.89187e-06, gain ratio: 1.61017e-02
[2024-07-26 14:11:26.654] [info] LM<sym::Optimize> [iter    9] lambda: 1.000e-01, error prev/linear/new: 2.009e+02/2.009e+02/2.011e+02, rel reduction: -6.77233e-04, gain ratio: -2.01352e+00
[2024-07-26 14:11:26.656] [info] LM<sym::Optimize> [iter   10] lambda: 1.000e+00, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: -4.29621e-06, gain ratio: -4.41172e-02
[2024-07-26 14:11:26.659] [info] LM<sym::Optimize> [iter   11] lambda: 1.000e+01, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: 1.68324e-05, gain ratio: 1.27198e+00
[2024-07-26 14:11:26.661] [info] LM<sym::Optimize> [iter   12] lambda: 1.000e+00, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: -4.27360e-06, gain ratio: -4.47309e-02
[2024-07-26 14:11:26.664] [info] LM<sym::Optimize> [iter   13] lambda: 1.000e+01, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: 1.65771e-05, gain ratio: 1.28204e+00
[2024-07-26 14:11:26.666] [info] LM<sym::Optimize> [iter   14] lambda: 1.000e+00, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: -4.21685e-06, gain ratio: -4.49676e-02
[2024-07-26 14:11:26.668] [info] LM<sym::Optimize> [iter   15] lambda: 1.000e+01, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: 1.63717e-05, gain ratio: 1.29433e+00
[2024-07-26 14:11:26.671] [info] LM<sym::Optimize> [iter   16] lambda: 1.000e+00, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: -4.13596e-06, gain ratio: -4.49176e-02
[2024-07-26 14:11:26.673] [info] LM<sym::Optimize> [iter   17] lambda: 1.000e+01, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: 1.62052e-05, gain ratio: 1.30846e+00
[2024-07-26 14:11:26.676] [info] LM<sym::Optimize> [iter   18] lambda: 1.000e+00, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: -4.03752e-06, gain ratio: -4.46405e-02
[2024-07-26 14:11:26.678] [info] LM<sym::Optimize> [iter   19] lambda: 1.000e+01, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: 1.60693e-05, gain ratio: 1.32411e+00
[2024-07-26 14:11:26.681] [info] LM<sym::Optimize> [iter   20] lambda: 1.000e+00, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: -3.92577e-06, gain ratio: -4.41744e-02
[2024-07-26 14:11:26.683] [info] LM<sym::Optimize> [iter   21] lambda: 1.000e+01, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: 1.59577e-05, gain ratio: 1.34101e+00
[2024-07-26 14:11:26.685] [info] LM<sym::Optimize> [iter   22] lambda: 1.000e+00, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: -3.80335e-06, gain ratio: -4.35422e-02
[2024-07-26 14:11:26.687] [info] LM<sym::Optimize> [iter   23] lambda: 1.000e+01, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: 1.58654e-05, gain ratio: 1.35893e+00
[2024-07-26 14:11:26.689] [info] LM<sym::Optimize> [iter   24] lambda: 1.000e+00, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: -3.67189e-06, gain ratio: -4.27562e-02
[2024-07-26 14:11:26.691] [info] LM<sym::Optimize> [iter   25] lambda: 1.000e+01, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: 1.57885e-05, gain ratio: 1.37768e+00
[2024-07-26 14:11:26.693] [info] LM<sym::Optimize> [iter   26] lambda: 1.000e+00, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: -3.53228e-06, gain ratio: -4.18221e-02
[2024-07-26 14:11:26.696] [info] LM<sym::Optimize> [iter   27] lambda: 1.000e+01, error prev/linear/new: 2.009e+02/2.009e+02/2.009e+02, rel reduction: 1.57240e-05, gain ratio: 1.39712e+00
[... cut ...]
3 might make sense given this - the dense cholesky solver SymForce uses is Eigen's LDLT, which has pivoting, which may make it more stable in the face of the hessian being extremely ill-conditioned.

When you say this is VINS-MONO, is it the default implementation https://github.com/HKUST-Aerial-Robotics/VINS-Mono, does it have modifications, is it another implementation, etc?  Partly I'm wondering whether your initializations are already very good, and the optimizations you're doing here are making things much worse than just the initialization, and what it would look like if the optimization just did nothing

empty-spacebar, 2024-07-27:
Here's my further test result, all the test is runing on the euroc MH_05_difficult dataset, using Symforce with DYNAMIC lambda parameter.
[0727.zip](https://github.com/user-attachments/files/16399017/0727.zip)

To facilitate your understanding, I make a table:
[image]

And here's some new conclusion:
1. The value of initial_lambda doesn't make many difference;
2. Min lambda in 1e-10 is reasonable, the smaller the worse, and too small(like 1e-16) lead to system crush;
3. It's interesting that I got the fastest runtime performance when I use sparse mode(see the yellow line in the table), it's also the first time symforce runs in real-time.

> Partly I'm wondering whether your initializations are already very good

Both Ceres and Symforce use the same initilization result, I didn't modify the initiation part of VINS-MONO.

empty-spacebar, 2024-08-09:
If you need , I can provide modified VINS-mono
Keep me updated.

aaron-skydio, 2024-08-09:
To be honest, it's unlikely that I'm going to have time to look into this soon.  Assuming you're talking about a fork of VINS-Mono, running the way they describe in their README for EuRoC on MH_05_difficult with nothing else involved, yeah it seems like a link to the code should be enough to reproduce, that would be helpful if I or someone else has time to look at some point

empty-spacebar, 2024-08-09:
Leave your email or any other contact information plz, or you can contact me by [zhangda12138@gmail.com]，then I could send you the code.

aaron-skydio, 2024-08-09:
It's going to be much easier if you make a fork of VINS-Mono, and link to your fork here.  If for some reason you don't want your code to be publicly accessible, you can create a private repo and invite my GitHub account.

empty-spacebar, 2024-08-12:
https://github.com/wxliu1/VINS-Mono-NotedByLWX
Here's the code

ZhouJankin, 2024-12-05:
@empty-spacebar Hi, I am also using SymForce to handle optimization problems, and I have encountered an issue where the optimization iteration time in C++ is too long, approximately  7ms per iteration. May I ask how long each iteration takes in the SymForce implementation you modified for VINS? Do you have any relevant data that you can share?

empty-spacebar, 2024-12-06:
> @empty-spacebar Hi, I am also using SymForce to handle optimization problems, and I have encountered an issue where the optimization iteration time in C++ is too long, approximately 7ms per [...]
Comparison between different hardware or different problem makes no sense, while comparison between solver makes sense. 

When solving BAL problem, symforce literally does a better job then Ceres.

closed 2025-09-30 by empty-spacebar