symforce-org/symforce #400: Build fails due to spaces in directory paths

Aterolite24, 2024-09-20:
### **Describe the bug**
When building the project in a directory path that contains spaces, the build process fails due to improper handling of spaces in directory paths. The path is interpreted incorrectly, causing the build to stop with errors related to path formatting.

For example, a directory like `C:/Users/Username/My Projects/Symforce` causes build failures, as spaces are not properly escaped or quoted.

### **To Reproduce**
Steps to reproduce the behavior:
1. Clone the repository into a directory that contains spaces, e.g., `C:/Users/Username/My Projects/Symforce`.
2. Attempt to build the project with the following command:
   ```bash
   python setup.py build
   # or 
   make px4_sitl_default none_iris
   ```
3. Observe the build failure due to improper path handling.

### **Expected behavior**
The build process should correctly handle directory paths with spaces by escaping or quoting the paths properly, allowing the build to proceed without errors.

### **Screenshots**
N/A (Optional if you want to include terminal screenshots showing the error.)

### **Environment (please complete the following information):**
- **OS and version:** Windows 10
- **Python version:** 3.11
- **SymForce Version:** (please specify the version or commit hash you're using)

Aterolite24, 2024-09-20:
#401 gives my approach to solve the problem. Thanks.

aaron-skydio, 2024-09-23:
Thanks for reporting - generally building on Windows isn't something we support or test in CI, but it sounds like this is probably a cross-platform problem anyway.  Building by invoking `setup.py` directly is also [deprecated by setuptools](https://packaging.python.org/en/latest/discussions/setup-py-deprecated/) and not supported, but that's also probably not the issue.

Do you have the output of an example build failure?  If not we'll see if we can reproduce.

Aterolite24, 2024-09-23:
Ahh I actually changed my setup.py since then .. and the issue was solved
actually so .. 😅

closed 2024-12-24 by aaron-skydio