symforce-org/symforce #404: Questions about how to specify the optimizer

zhanghua7099, 2024-10-27:
Hi! I have a question about how to specify the optimizer. In the tutorial, 
https://github.com/symforce-org/symforce/tree/main?tab=readme-ov-file#solve-the-problem
The **LM algorithm** seems to be the **default** method for solving the factor graph. 

I also found that this project seems to support the **GNC solver**. 
https://symforce.org/api-cpp/class/classsym_1_1GncOptimizer.html

Could you please give me some advice about how to use GNC solver? A simple demo will be OK. 

Thank you very much!

aaron-skydio, 2024-10-27:
The GNC optimizer currently isn't exposed through the Python bindings.  You can use it from C++ though, see [symforce_gnc_test](https://github.com/symforce-org/symforce/blob/04ccac19aa25bf657c1f312539fc6964fd402128/test/symforce_gnc_test.cc) for an example.

Levenberg Marquardt is still used as the nonlinear solver when using the GNC optimizer; which nonlinear optimizer to use is a separate choice from whether or not to use GNC.  It's a template parameter to the Optimizer or the GncOptimizer, although the only provided nonlinear solver is LM.

zhanghua7099, 2024-10-28:
@aaron-skydio Hi! Thanks for your reply. I found a codegen demo using Python. 
https://github.com/symforce-org/symforce/blob/04ccac19aa25bf657c1f312539fc6964fd402128/test/symforce_gnc_codegen_test.py#L29-L32
I have another question about the use of the `noise_models`. Does the code `noise_model.whiten(x - y)` mean that the factor adds the robust kernel? (If I'm right, `x - y` will be the ordinary least squares.)

The second question is 
https://github.com/symforce-org/symforce/blob/main/README.md#build-an-optimization-problem

When adding the factors, 
```
factors = []

# Bearing factors
for i in range(num_poses):
    for j in range(num_landmarks):
        factors.append(Factor(
            residual=bearing_residual,
            keys=[f"poses[{i}]", f"landmarks[{j}]", f"angles[{i}][{j}]", "epsilon"],
        ))
```
The speed is very slow if the number of landmarks is large. Do you have some suggestions about speeding ​​up this operation?

Thank you for your help!

aaron-skydio, 2024-10-29:
> I have another question about the use of the `noise_models`. Does the code `noise_model.whiten(x - y)` mean that the factor adds the robust kernel? (If I'm right, `x - y` will be the ordinary least squares.)

Yes that's right.  There's no magic in the `Factor`, all it sees is a symbolic residual (which may include modifications like a robust kernel).

> The speed is very slow if the number of landmarks is large. Do you have some suggestions about speeding ​​up this operation?

How many landmarks?  How many poses?  How slow?  Are you sure the time is spent in that block of code exactly?  It looks like that loop takes around 10s for me with 10000 landmarks (and 3 poses), or 2s with 1000 landmarks

zhanghua7099, 2024-10-29:
> How many landmarks? How many poses? How slow? Are you sure the time is spent in that block of code exactly? It looks like that loop takes around 10s for me with 10000 landmarks (and 3 poses), or 2s with 1000 landmarks.

Hi! Thanks for your reply! The results on my computer are similar (around 10s with 10000 landmarks). Is there a better way to speed up the process?

aaron-skydio, 2024-10-29:
You might be able to try creating one `Factor` with some dummy key names, converting that to a `NumericFactor` with `Factor.to_numeric_factor`, and then just creating a `NumericFactor` with the same `linearization_function` and the right keys for each pose-landmark pair, which may be faster.  The actual codegen is already cached for creating multiple `Factor` objects, but you'll skip the actual symbolic function invokation, which may be expensive.

zhanghua7099, 2024-10-29:
Hi! @aaron-skydio 

Thank you very much for your help! After using `NumericFactor`, the code runs fast (**10 s -> less than 0.1 s**). 

I use the following steps to realize this. 

1. Wirte the `residual_function`
2. Use`Codegen` to generate the `linearization_function` of `residual_function`
3. Use the `symforce.opt.factor.NumericFactor` to replace `from symforce.opt.factor import Factor`
4. Solve the problem

The solved results are the same. Thanks again for your help!

closed 2024-10-29 by zhanghua7099