symforce-org/symforce #406: filename has 'skymarshal', but metadata has 'unknown'

digiamm, 2024-11-18:
I am encountering this error when installing via pip - `filename has 'skymarshal', but metadata has 'unknown'`. 

To isolate I have created a minimal docker with Ubuntu 22.04

```bash
apt install python3-pip
apt install libgmp-dev
pip install .
```

Here the full output:
```bash
root@8af5b39e5ecf:/deps/symforce# pip install .
Processing /deps/symforce
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install backend dependencies did not run successfully.
  │ exit code: 1
  ╰─> [13 lines of output]
      Processing //deps/symforce/third_party/skymarshal
        Installing build dependencies: started
        Installing build dependencies: finished with status 'done'
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'done'
        Installing backend dependencies: started
        Installing backend dependencies: finished with status 'done'
        Preparing metadata (pyproject.toml): started
        Preparing metadata (pyproject.toml): finished with status 'done'
        WARNING: Generating metadata for package skymarshal produced metadata for project name unknown. Fix your #egg=skymarshal fragments.
      Discarding file://localhost//deps/symforce/third_party/skymarshal: Requested unknown from file://localhost//deps/symforce/third_party/skymarshal has inconsistent name: filename has 'skymarshal', but metadata has 'unknown'
      ERROR: Could not find a version that satisfies the requirement skymarshal (unavailable) (from versions: 0.3.dev2, 0.3.dev3, 0.3.dev5, 0.4.0, 0.5.0, 0.6.0, 0.7.0, 0.8.0, 0.9.0)
      ERROR: No matching distribution found for skymarshal (unavailable)
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install backend dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

**Environment:**
 - OS and version: Ubuntu 22.04
 - Python version 3.10.12
 - SymForce Version latest from source

aaron-skydio, 2024-11-18:
What versions of `pip` and `setuptools` are you using?

aaron-skydio, 2024-11-18:
To answer my own question, probably:
```
root@994841b34b49:/# pip --version
pip 22.0.2 from /usr/lib/python3/dist-packages/pip (python 3.10)
root@994841b34b49:/# python3 --version
Python 3.10.12
root@994841b34b49:/# python3 -m pip show setuptools
Name: setuptools
Version: 59.6.0
Summary: Easily download, build, install, upgrade, and uninstall Python packages
Home-page: https://github.com/pypa/setuptools
Author: Python Packaging Authority
Author-email: distutils-sig@python.org
License: UNKNOWN
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: 
```

in docker `ubuntu:jammy` hash `sha256:0e5e4a57c2499249aafc3b40fcd541e9a456aab7296681a3994d631587203f97`

digiamm, 2024-11-19:
Thanks, everything is the same but I did not specify that the docker image I am using is Ubuntu 22.04 with CUDA drivers: `nvidia/cuda:12.6.2-devel-ubuntu22.04`. The digest is: `sha256:1c253f721b28b27c924677493c8ed45812c1b1a562bca828c144683b444e7298`.

aaron-skydio, 2024-12-21:
I believe this is this pip / debian / ubuntu jammy bug: https://github.com/pypa/setuptools/issues/3269

So not a SymForce bug, and should be resolved by updating pip (or doing a non-system install etc).  The install succeeds if I upgrade pip to 23.3.2.  I think according to that thread 22.3 should resolve, but I was still running into issues on 22.3.

I.e. on a fresh `ubuntu:jammy` container, this sequence succeeds:

```
apt update
apt install python3-pip libgmp-dev git
mkdir deps
cd deps
git clone https://github.com/symforce-org/symforce
cd symforce
pip install pip~=23.0
pip install .
```

closed 2024-12-21 by aaron-skydio