symforce-org/symforce #407: Schur Complement for Bundle Adjustment

MGBzH, 2024-11-21:
Hi, first of all thanks for your amazing work!

I'm using symforce to solve classical bundle adjustment with 6 DoF poses, 3D points and projection factors. The optimization works as expected, however is the Schur Complement trick used automatically or do I need to do something to enable it to speed up the optimisation?

Thank you

aaron-skydio, 2024-11-21:
The default configuration uses METIS to compute the ordering used for elimination.  The resulting orderings have the same asymptotic complexity as the Schur complement trick (linear in the number of 3D points in the problem), so it should be pretty fast and it'll scale as you expect with the numbers of poses and points, same as if you were explicitly using the Schur complement trick

For really big problems there are other tricks you can use and lots of linear solvers to choose from; the linear solver is a template parameter that you can change to any solver that implements the interface SymForce expects for linear solvers; for example, you can use any of Eigen's sparse linear solvers with the wrapper in [eigen_sparse_solver.h](https://github.com/symforce-org/symforce/blob/04ccac19aa25bf657c1f312539fc6964fd402128/symforce/opt/eigen_sparse_solver.h), or you could plug in a solver from another library by implementing a similar adapter.

MGBzH, 2024-11-22:
Thank you for your answer.
I am working on a relatively small BA problem (35 poses, 2300 points and around 19k edges). 
To understand how code generation works I've generated a reprojection residual. I've modified the `bundle_adjustment_in_the_large.py` example by creating a reprojection residual between a `sf.Pose3` and a `sf.V3` but using a pinhole camera model with `sf.LinearCameraCal`. Those intrinsics are known and fixed so I did not include them in the `which_args` parameter list of Codegen.
I have then built a graph with the generated factor, following the implementation of `bundle_adjustment_in_the_large.cc` but using my own data and optimized using sym::Optimizerd with default params and fixed the first keyframe of the graph.
I did a similar approch but using g2o (with analytical jacobians and the Schur complement trick) and I get to the same optimum as with symforce. 
However while symforce converges in fewer (6) iterations than g2o (10), the time per iteration of g2o is lower: around 0.02-0.04s against 0.1 for symforce, making the overall optimisation take 0.2s for g2o against 0.6 for symforce.

Have you been able to compare against g2o in your experiments and is this difference something you have observed? Or on the contrary have you seen that symforce was faster, suggesting an error on my side?

Thank you

aaron-skydio, 2024-11-23:
It would help if you could post more detailed timing for what part of the iteration time is slower; there are very different reasons depending on that.  SymForce should print a table on exit of timing for different parts of the optimization problem; I haven't used g2o much so I'm not sure what timing information they have available

MGBzH, 2024-12-02:
Thanks for your answer, here are the detailed timings for symforce on my BA graph
```
SymForce TicToc Results:
   Name                                                          :     Count      | Total Time (s) | Mean Time (s)  |  Max Time (s)  |  Min Time (s) 
-----------------------------------------------------------------+----------------+----------------+----------------+----------------+----------------
Optimizer<BundleAdjustmentOptimizer>::Optimize                   :       1        |    0.58763     |    0.58763     |    0.58763     |    0.58763    
Optimizer<BundleAdjustmentOptimizer>::IterateToConvergence       :       1        |    0.58437     |    0.58437     |    0.58437     |    0.58437    
LM<BundleAdjustmentOptimizer>::Iterate()                         :       6        |    0.58357     |    0.097262    |    0.21112     |    0.072172   
LM<BundleAdjustmentOptimizer>: SparseFactorize                   :       6        |    0.35627     |    0.059378    |    0.061376    |    0.057262   
LM<BundleAdjustmentOptimizer>: EvaluateFirst                     :       1        |    0.091917    |    0.091917    |    0.091917    |    0.091917   
Linearizer<BundleAdjustmentOptimizer>::Relinearize::First()      :       1        |    0.091861    |    0.091861    |    0.091861    |    0.091861   
Linearizer<BundleAdjustmentOptimizer>::Relinearize::NonFirst()   :       6        |    0.078893    |    0.013149    |    0.013604    |    0.012917   
LM<BundleAdjustmentOptimizer>: AnalyzePattern                    :       1        |    0.041681    |    0.041681    |    0.041681    |    0.041681   
LM<BundleAdjustmentOptimizer>: SparseSolve                       :       6        |   0.0088856    |   0.0014809    |   0.0015008    |   0.0014434   
LM<BundleAdjustmentOptimizer>: Update                            :       6        |   0.0035052    |   0.00058419   |    0.00285     |   0.00012218  
LM<BundleAdjustmentOptimizer>::ResetState                        :       1        |   0.0020572    |   0.0020572    |   0.0020572    |   0.0020572   
LM<BundleAdjustmentOptimizer>: DampHessian                       :       6        |   0.00079118   |   0.00013186   |   0.00013784   |   0.00012834  
Optimizer<BundleAdjustmentOptimizer>::CopyValuesAndLinearization :       1        |    0.000788    |    0.000788    |    0.000788    |    0.000788   
LM<BundleAdjustmentOptimizer>: ResetHessianDiagonal              :       6        |   0.00050138   |   8.3563e-05   |   8.5391e-05   |   8.166e-05   
LM<BundleAdjustmentOptimizer>: IterationStats                    :       6        |   0.00038089   |   6.3482e-05   |   0.00015885   |   4.3121e-05  
LM<BundleAdjustmentOptimizer>: IterationStats - Print            :       6        |   0.0003704    |   6.1734e-05   |   0.00015399   |   4.1811e-05  
LM<BundleAdjustmentOptimizer>: LinearErrorFromValues             :       6        |   5.859e-05    |   9.765e-06    |   1.201e-05    |    8.63e-06   
LM<BundleAdjustmentOptimizer>: accept_update bookkeeping         :       6        |   3.577e-05    |   5.9617e-06   |    9.45e-06    |    5.06e-06   
LM<BundleAdjustmentOptimizer>: FirstIterationStats               :       1        |   2.847e-05    |   2.847e-05    |   2.847e-05    |   2.847e-05   
LM<BundleAdjustmentOptimizer>: StateStep                         :       6        |    1.78e-06    |   2.9667e-07   |    1.34e-06    |     8e-08
```
with the following cost changes:
```
LM<BundleAdjustmentOptimizer> [iter    0] lambda: 1.000e+00, error prev/linear/new: 1.155e+06/1.054e+04/1.745e+05, rel reduction: 8.48895e-01, gain ratio: 8.56718e-01
LM<BundleAdjustmentOptimizer> [iter    1] lambda: 6.369e-01, error prev/linear/new: 1.745e+05/1.442e+04/2.998e+04, rel reduction: 8.28198e-01, gain ratio: 9.02807e-01
LM<BundleAdjustmentOptimizer> [iter    2] lambda: 3.039e-01, error prev/linear/new: 2.998e+04/2.339e+04/2.709e+04, rel reduction: 9.62094e-02, gain ratio: 4.37668e-01
LM<BundleAdjustmentOptimizer> [iter    3] lambda: 3.045e-01, error prev/linear/new: 2.709e+04/2.328e+04/2.338e+04, rel reduction: 1.36873e-01, gain ratio: 9.73054e-01
LM<BundleAdjustmentOptimizer> [iter    4] lambda: 1.015e-01, error prev/linear/new: 2.338e+04/2.328e+04/2.329e+04, rel reduction: 3.85861e-03, gain ratio: 8.35394e-01
LM<BundleAdjustmentOptimizer> [iter    5] lambda: 7.086e-02, error prev/linear/new: 2.329e+04/2.328e+04/2.329e+04, rel reduction: 2.46218e-04, gain ratio: 5.07957e-01
LM<BundleAdjustmentOptimizer> Optimization finished with status: SUCCESS
```
Sadly g2o does not provide detailed timings like symforce, only time per iteration:
```
iteration= 0     chi2= 349229.953631     time= 0.041154          cumTime= 0.041154       edges= 18559    schur= 1        lambda= 1462.611335     levenbergIter= 1
iteration= 1     chi2= 59875.579867      time= 0.0209294         cumTime= 0.0620834      edges= 18559    schur= 1        lambda= 693.552774      levenbergIter= 1
iteration= 2     chi2= 49773.811349      time= 0.0204836         cumTime= 0.082567       edges= 18559    schur= 1        lambda= 355.477057      levenbergIter= 1
iteration= 3     chi2= 48134.734576      time= 0.020891          cumTime= 0.103458       edges= 18559    schur= 1        lambda= 236.984705      levenbergIter= 1
iteration= 4     chi2= 47323.525489      time= 0.022424          cumTime= 0.125882       edges= 18559    schur= 1        lambda= 78.994902       levenbergIter= 1
[... cut ...] mple.cpp#L143)), I would expect a speed up from the implementation.
On problem 39 g2o converges to 121564 in 3.63s while symforce converges to 62160 in 8.63s.
Looking at the detailed timings for symforce it seems that SparseFactorize is taking a large amount of time compared to the whole convergence time of g2o. Is it possible that using a different factorization algorithm could explain this difference? G2o is using CHOLMOD, what is it possible to use with symforce?
Thank you again for your time

aaron-skydio, 2024-12-21:
So afaict the difference is that g2o has a specialized schur complement implementation, see here: https://github.com/RainerKuemmerle/g2o/blob/eec325a1da1273e87bc97887d49e70570f28570c/g2o/core/block_solver.hpp

Best I can tell, their use of CHOLMOD is just for doing sparse cholesky, and CHOLMOD's sparse cholesky should have very similar performance to SymForce's (CHOLMOD also has other solvers, but I believe sparse cholesky is the one we're talking about).

In this case though, they're forming the schur complement with a specialized implementation, which will have essentially the same asymptotic complexity as SymForce using METIS and our sparse cholesky, but will be some amount faster in practice (I'm not sure how much exactly).  They also support doing this in parallel with OpenMP, not sure if that's enabled in your tests, but that'll also make things faster.

It might be interesting to add an explicit schur complement solver to SymForce - we actually have one already in `symforce/opt/sparse_schur_solver.h`, it just doesn't quite conform to the API expected by our nonlinear optimizer, so it would be a little bit of effort to plug it in there (we currently use it for other things).