symforce-org/symforce #428: Lazy codegen spits void c++ function

JzHuai0108, 2025-02-27:
Here is my python code to generate the cpp functions:
```python
def nubspline_matrix(k: int, knots: sf.M, so: int, epsilon: sf.Scalar) -> sf.M:
    """
    Computes the M matrix recursively.
    :param k: Spline index
    :param knots: List or deque of knot values of len 2k
    :param so: spline order
    :return: M matrix
    """
    assert knots.shape[0] == 2 * so
    i = so - 1
    if k == 1:
        M = sf.M.ones(1, 1)
        return M
    else:
        M_km1 = nubspline_matrix(k - 1, knots, so, epsilon)
        # The recursive equation for M
        # M_k = [ M_km1 ] A  + [  0^T  ] B
        #       [  0^T  ]      [ M_km1 ]
        #        -------        -------
        #         =: M1          =: M2
        #     = M1 A + M2 B
        M1 = sf.M.zeros(M_km1.shape[0] + 1, M_km1.shape[1])
        M2 = sf.M.zeros(M_km1.shape[0] + 1, M_km1.shape[1])

        M1[:M_km1.shape[0], :] = M_km1
        M2[1:, :] = M_km1

        A = sf.M.zeros(k - 1, k)
        for idx in range(A.shape[0]):
            j = i - k + 2 + idx
            d0 = d_0(k, i, j, knots, epsilon)
            A[idx, idx] = 1.0 - d0
            A[idx, idx + 1] = d0

        B = sf.M.zeros(k - 1, k)
        for idx in range(B.shape[0]):
            j = i - k + 2 + idx
            d1 = d_1(k, i, j, knots, epsilon)
            B[idx, idx] = -d1
            B[idx, idx + 1] = d1

        M_k = M1 * A + M2 * B
        return M_k

def d_0(k: int, i: int, j: int, knots: sf.M, epsilon: sf.Scalar):
    """
    Compute the d_0 term for a specific spline.
    """
    denom = compute_duration(knots[j], knots[j + k - 1])
    if denom <= 0.0:
        return 0.0
    numerator = compute_duration(knots[j], knots[i])
    return divide_durations(numerator, denom, epsilon)

def d_1(k: int, i: int, j: int, knots: sf.M, epsilon: sf.Scalar):
    """
    Compute the d_1 term for a specific spline.
    """
    denom = compute_duration(knots[j], knots[j + k - 1])
    if denom <= 0.0:
        return 0.0
    numerator = compute_duration(knots[i], knots[i + 1])
    return divide_durations(numerator, denom, epsilon)

def compute_duration(start: sf.Scalar, end: sf.Scalar):
    return end - start

def divide_durations(numerator: sf.Scalar, denominator: sf.Scalar, epsilon: sf.Scalar):
    return numerator / (denominator + epsilon)

class Vector10(sf.Matrix):
    SHAPE = (10, 1)

def nubspline5_matrix(knots: Vector10, epsilon: sf.Scalar) -> sf.M:
    """
    Warn: The derivative is relative to ut instead of t.
    """
    n = 5
    return nubspline_matrix(n, knots, n, epsilon).transpose()

from symforce.codegen import Codegen
from symforce.codegen import CppConfig
from functools import partial

output_dir = "/home/pi/Desktop/temp"
print_code = True

codegen = Codegen.function(nubspline5_matrix, config=CppConfig())
metadata = codegen.generate_function(output_dir=output_dir, skip_directory_nesting=True)
if print_code:
    print(metadata.generated_files[0].read_text())

n = 5
knots = sf.M(n * 2, 1)
dt = 0.01
for i in range(n * 2):
    knots[i] = 100 + i * dt
m5 = nubspline5_matrix(knots, epsilon=0)
print(f'24 m5: {m5 * 24}')
```
The generated code looks like

```cpp
#pragma once

#include <Eigen/Dense>

namespace sym {

/**
 * Warn: The derivative is relative to ut instead of t.
 */
template <typename Scalar>
Eigen::Matrix<Scalar, 5, 5> Nubspline5Matrix(const Eigen::Matrix<Scalar, 10, 1>& knots,
                                             const Scalar epsilon) {
  // Total ops: 0

  // Unused inputs
  (void)knots;
  (void)epsilon;

  // Input arrays

  // Intermediate terms (0)

  // Output terms (1)
  Eigen::Matrix<Scalar, 5, 5> _res;

  _res.setZero();

  _res(0, 0) = Scalar(1.0);

  return _res;
}  // NOLINT(readability/fn_size)

// NOLINTNEXTLINE(readability/fn_size)
}  // namespace sym
```
But the function nubspline5_matrix works well in python.
Can you please help me understand the issue? I think I have misused some features.

JzHuai0108, 2025-02-28:
Removing the two if's in d_0 and d_1 will make the codegen work.

aaron-skydio, 2025-02-28:
Is this resolved then?

JzHuai0108, 2025-02-28:
Yes. I did unit tests on the generated cpp code. It produces expected results.

closed 2025-02-28 by JzHuai0108

JzHuai0108, 2025-02-28:
BTW, do you think the epsilon in 
```
def divide_durations(numerator: sf.Scalar, denominator: sf.Scalar, epsilon: sf.Scalar):
    return numerator / (denominator + epsilon)
```
needed?

aaron-skydio, 2025-02-28:
It's not needed as long as the difference between knot points is guaranteed to be nonzero (or not close to zero), e.g. if you were optimizing the knot points or something you might want the epsilon, but if they're known to have constant nonzero spacing you don't need the epsilon

JzHuai0108, 2025-02-28:
I agree. Thank you for the clarification.