symforce-org/symforce #430: Mysterious behaviors of code generated with symforce

JzHuai0108, 2025-03-10:
Dear authors,

I am using symforce to generate some cost functions, and use them in an optimization problem solved by the  ceres solver and our custom solver.
The developing env is Ubuntu 22 + ROS2 Humble.

There are quite a few strange things I encountered. 
If you have experience with these, you may hint me how to fix them. 
I'd be happy to share my codebase with you if you are interested to take a look.
Meanwhile, I will take time to pinpoint the cause by unit tests as I find time.

The first warning sign is that the optimizer from ceres solver or our solver does not converge for the problem built with sym jacobians. For the sparse normal cholesky solver in ceres solver, the optimized parameters barely change from their initial values at all. Our solver can change these parameters torwards the reference value more or less. 
However, the same problem with my analytic Jacobians and my custom parameter types can be solved by the ceres solver and our custom solver with good convergence.

So I check the Jacobians of all cost functions with numeric Jacobians.
I found the numeric Jacobians sometimes do not agree with symbolic Jacobians. For a parameter, its numeric Jacobian in some cost functions differs very much from the value given by symforce. 
This problem occurs mostly with the rotation part of a pose.
However, the two Jacobians can agree well in some cases of the same type of the cost function.

In the same terminal, I got very different results when I ran the same program with absolute path and with relative path.
The case with relative path gives lots of wrong residuals containing inf values.
The case with absolute path looks fine in terms of residuals.
But both cases do not converge.
This erratic behavior has not been observed with my other program that solves the same opt problem.

Now I have doubts about several points.
1. Do we need EIGEN_MAKE_ALIGNED_OPERATOR_NEW in C++20 programs?
2. Since symforce may not generate code properly for if else branching, does it work with for loops which iterates an integer range?

aaron-skydio, 2025-03-10:
A.  It's going to be pretty hard to help with this without more information.  But, you mention it's mostly with the rotation part of a pose - have you checked that the tangent space SymForce is using is consistent with the tangent space your optimizer is using, and consistent with the tangent space you're using to perturb variables for numerical derivatives?
C.  No, this should not be necessary in C++17 onward: https://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html
D. It's common to have loops in symbolic functions.  The way SymForce figures out what to generate from your function isn't magical - all that it does is call your function once, with arguments matching exactly the types you've annotated.  Your function returns one or more symbolic expressions that SymForce then generates code for.  This means whatever code path is taken through your function when it's called by `Codegen.function` is what you'll get; if you add printouts to your function for example, you should easily be able to see what branches are taken or how many iterations of loops are executed, or print out intermediate expressions.  The code for this is [here](https://github.com/symforce-org/symforce/blob/da8ea1a87a9b541047c57bd326bbf8868adc604c/symforce/codegen/codegen.py#L256-L347).

JzHuai0108, 2025-03-13:
A. I will dig into the problem with your hints.
C. Thank you for the confirmation.
D. Now I largely understand what happens under the hood.

JzHuai0108, 2025-03-13:
I checked every Jacobian of every cost function by comparing the numeric diff and the symbolic diff values. They agree pretty well, usually with difference less than 1e-5.
The optimizer reports convergence, and some parameters move to the expected values. But many parameters just stay unchanged as before.

JzHuai0108, 2025-03-15:
So I changed the critical_theta to 2 pi for the Spherical camera model, the symptom does not change.
Then I enlarged the robust loss window width to very large values, again the symptom does not change.
The symptom so far is that the camera extrinsic parameters and time offset parameter does not change at all after many iterations of optimization regardless of the camera model and the optimization solver.